msm8916-mainline/lk2nd #28: Support for Samsung Galaxy Ace Style SM-G357FZ

arthurlutz, 2020-08-16:
According to the wikipedia page https://en.wikipedia.org/wiki/Samsung_Galaxy_Ace_4 the SM-G357FZ uses Snapdragon 410 MSM8916. 

Any chance of getting lk2nd working with it ? (and by extension postmarketos). I have one at hand and would be able to test it out / debug stuff.

Minecrell, 2020-08-16:
If you have MSM8916 (Snapdragon 410) there is a good chance that you can get something working.

Please take a look at this wiki page: https://wiki.postmarketos.org/wiki/MSM8916_Mainlining
and join the [postmarketOS mainlining channel](https://wiki.postmarketos.org/wiki/Matrix_and_IRC) for help (available via IRC or Matrix).

Minecrell, 2020-10-16:
Please let us know if you need any further help, I will close this for now :)

closed 2020-10-16 by Minecrell