msm8916-mainline/lk2nd #35: htc-a51tuhl: dtbTool need add some extra item for htc

Hinokami-Kagura, 2020-10-08:
After checked the dtb generated from stock dts, found it ignores the '-d "htc,project-id = <"', cause the dtb is lost, from HTC fastboot oem demsg :

INFO[6646622] fastboot_command:[oem dmesg]
INFO[6646636] fb_cmd_oem:dmesg
INFO[6646695] register_debug_console: new entry ptr=0x83000000
INFO[8782573] fastboot_command:[download:000a6010]
INFO[8782588] CMD:download: download:000a6010
INFO[8782665] recv data addr=86800000 size=A6010
INFO[8782677] status: DATA000a6010
INFO[8783007] Prepare to download data[679952 Bytes]...
INFO[8973761] fastboot_command:[boot]
INFO[8973800] [DEBUG] Board pid:379, pcbid:0x80, soc version:0x1
INFO0000 
INFO[8973817] [DEBUG] DTB entry[1] pid:239, pcbid:0x1, hw_subtyp
INFOe:0x0, soc version:0x0 (offset:0x800, size:0x33000)
INFO[8973833] [DEBUG] DTB entry[2] pid:241, pcbid:0x1, hw_subtyp
INFOe:0x0, soc version:0x0 (offset:0x800, size:0x33000)
INFO[8973848] [DEBUG] DTB entry[3] pid:263, pcbid:0x1, hw_subtyp
INFOe:0x0, soc version:0x0 (offset:0x800, size:0x33000)
INFO[8973861] [CRITICAL] ERROR: Getting device tree address fail
INFOed

is there can add that function in dtbTool? it's not like oppo need different dtbtool to build special dtb...

Minecrell, 2020-10-16:
I think this was solved by setting `htc,project-id = qcom,msm-id` :)

closed 2020-10-16 by Minecrell

Hinokami-Kagura, 2020-10-16:
take note about it :

for HTC phone which is S-ON status (if the phone is updated stock ROM to lollipop & Marshmallow with newer Hboot, it won't be S-Off easily) 
It need to Install the lk2nd to Recovery partition for skipping the Hboot validating process for boot partition & system partition.
seems it'll give hope for many HTC device which is msm8226/msm8916/msm8939 under S-on mode, thanks!