msm8916-mainline/lk2nd #64: lk2nd not working on Samsung S4 mini?

jhaar, 2021-04-05:
Hi there

I'm wanting to use postmarketos and so installed current lk2nd (on Ubuntu) via "heimdall flash --BOOT lk2nd.img". No sign of error occurred and the phone auto-rebooted. I see the Samsung boot logo, and then I held down the Power+VolDown keys, but nothing changes - no sign of lk2nd executing (also tried holding down keys before seeing the logo). I've also tried with just VolDown and even re-flashed with "--no-reboot" and did the reboot myself - no difference - and of course "fastboot flash ..." doesn't work as it's Samsung.

Any ideas what's wrong? The phone has TWRP and LineageOS on it - I don't know if that messes with anything - but I'll mention it for completeness. 

Thanks

Minecrell, 2021-04-05:
This version of lk2nd only works on the  Samsung Galaxy S4 Mini **Value Edition** (aka GT-I9195**I**). Do you actually have the correct one? :)

jhaar, 2021-04-05:
D'oh! Nope I have the International one :-(

I'll go try the pmbootstrap sideload option now

sorry about that :-)

closed 2021-04-05 by jhaar