msm8916-mainline/lk2nd #66: Not detected by PC (SM-G600FY)

alou-S, 2021-04-23:
The image flashed and booted fine and works.
But it is not detected by the PC.

`fastboot devices` on Linux gives no reply (I have tried running it as root)

The device doesn't pop up on Device Manager and well `fastboot devices` doesn't list it either.

Samsung On 7 Pro (SM-G600FY)
MSM-8916

jja2000, 2021-04-24:
Hi! What stock Android version were you running before you flashed lk2nd? And could you maybe upload a screenshot of what you see on the screen possibly with another phone?

alou-S, 2021-04-25:
It was Android 9 ( Lineage OS 16 ). Also how will Android versions effect lk2nd?

Anyway here's the image screenshot you requested 
[image]

jja2000, 2021-04-25:
I am not necessarily talking about Lineage OS, but Samsungs stock version installed on the G600FY.

This matters because the id used to detect what Devicetree should be used changed inbetween android 5 (the one my SM-G6000 ships with) and Android 6 (the one yours should ship with, I think Brazil got 7 too).
The fix for USB is applied per specific id, since mine is different to yours my usb works while yours doesn't.

@Minecrell is it wise to test if flashing my device to the G600FY works (like Sadiq's A6000) and then just merge G6000 with G600FY or should the G600FY get a seperate port for now?

alou-S, 2021-04-25:
Oh I see. Yes, my device shipped with 6.0

jja2000, 2021-04-26:
I would advise you to join #postmarketos-mainline on matrix and make a new port for the G600FY. It is incompatible with my G6000 it seems.

alou-S, 2021-04-26:
Sure thing. Thanks for the help.

alou-S, 2021-04-26:
Sure thing. Thanks for the advice.

closed 2021-04-26 by alou-S