msm8916-mainline/lk2nd #75: kiwi doesn't connect to mobile network

fynngodau, 2021-05-14:
When booting through lk2nd, my huawei-kiwi (Honor 5X, KIW-L21) doesn't connect to the mobile network anymore in my latest LineageOS system ("No service"). Everything is working whenever I boot the device from the stock bootloader.

To illustrate this further, here's my sequence of commands (but of course it is the same result when flashing lk2nd to boot instead of booting it directly):

```
$ adb reboot bootloader
$ fastboot boot build-msm8916-secondary/lk2nd.img # or an image from the releases download
Downloading 'boot.img'                             OKAY [  0.012s]
booting                                            OKAY [  0.002s]
Finished. Total time: 0.015s
$ fastboot boot ~/boot.img # original LineageOS boot image
Downloading 'boot.img'                             OKAY [  0.969s]
booting                                            OKAY [  0.102s]
Finished. Total time: 1.092s

# phone doesn't connect to mobile network now

$ adb reboot bootloader
$ fastboot boot ~/boot.img
Downloading 'boot.img'                             OKAY [  0.958s]
booting                                            OKAY [  0.017s]
Finished. Total time: 0.986s

# phone now connects to mobile network again
```

Hoping that it helps, here is an [lk_log.txt](https://github.com/msm8916-mainline/lk2nd/files/6478980/lk_log.txt).

Minecrell, 2021-05-14:
Well, I guess this is some Huawei mess :)

Minecrell, 2021-05-26:
I'm gonna close this issue. Booting downstream/vendor kernels is only a secondary goal for lk2nd. It should preferably work somewhat but I have no plans to investigate vendor-specific mess like this. Hope that's understandable :)

closed 2021-05-26 by Minecrell

fynngodau, 2021-05-26:
Okay, too bad that there's no simple solution.

Minecrell, 2021-05-26:
I'm guessing the Huawei bootloader may patch something into the DTB, which isn't done when booting with lk2nd. But it could be also anything else.

Newbytee, 2021-09-15:
@fynngodau can you try to add `huawei,modem_id = <0xFFB40105 0x0>;` to the kiwi dts and see if that resolves your issue?

fynngodau, 2021-09-15:
@Newbytee Yes, indeed, this does resolve my problem (network connects after booting through lk2nd). Awesome!