msm8916-mainline/lk2nd #84: Build steps - help needed

romaduck, 2021-07-15:
I will build this project, but I don't understand what tool chain actually is. Can You help me please?
I use Linux Manjaro 64bit, Linux Kernel 5.13.1-3. Which tool chain prefix I have to use? 
`make TOOLCHAIN_PREFIX=arm-none-eabi- MSM8916-secondary`
Cheers, Roman

Minecrell, 2021-07-16:
I would expect that Manjaro has the same package as Arch Linux for the ARM cross compiler, i.e. `pacman -S arm-none-eabi-gcc dtc` as suggested in the README. Then you can use the command as you suggested

```
$ make TOOLCHAIN_PREFIX=arm-none-eabi- msm8916-secondary
```

romaduck, 2021-07-16:
Thank You for Your answer. 
Yes, I did it exactly like You say, but I'm getting an error all the Time. 
`$ make TOOLCHAIN_PREFIX=arm-none-eabi- msm8916-secondary
make[1]: Entering directory '/mnt/data/home/user/projects/myprj/lk2nd'
including lib/debug lib/heap lib/libc lib/openssl
including lib/openssl/crypto
make/build.mk:69: arch//compile.mk: No such file or directory
make[1]: *** No rule to make target 'arch//compile.mk'.  Stop.
make[1]: Leaving directory '/mnt/data/home/user/projects/myprj/lk2nd'
make: *** [makefile:20: make-make] Error 2`
Do I have to config something else, some rules for my project? 
So far, I created just one file - local.mk in the project root folder, with one singel row in it: `PROJECT := MSM8916`.

Minecrell, 2021-07-16:
Uh, don't create any files. Make a clean clone of lk2nd and build it exactly with the commands I mentioned. Nothing else.

romaduck, 2021-07-16:
Yes, it works now, thank You for Your Help! 
My CLI command was wrong, it has to be
`make TOOLCHAIN_PREFIX=arm-none-eabi- msm8916-secondary` not MSM8916 like in my first message. 
This toolchain prefix is case sensitive, I didn't know this. And I removed this local.mk file.

closed 2021-07-16 by romaduck