msm8916-mainline/lk2nd #105: lge-hammerhead: Cannot boot pmOS or TWRP on D820/D821(H)

wonderfulShrineMaidenOfParadise, 2021-10-06:
There are at least 4 hammehead models: D820/D821(E) and D820/D821(H).
E stands for ``elpida`` and H stands for ``hynix``.
pmOS and TWRP are able to boot on ``androidboot.hardware.ddr=elpida`` devices with the device tree.
But not on ``androidboot.hardware.ddr=hynix" devices``.
[lk_log-d820(h).txt](https://github.com/msm8916-mainline/lk2nd/files/7293509/lk_log-d820.h.txt)
[lk_log-d821(h).txt](https://github.com/msm8916-mainline/lk2nd/files/7293510/lk_log-d821.h.txt)

Minecrell, 2021-10-06:
I don't really understand this, perhaps an UART log or at least a detailed description of what happens when booting pmOS or TWRP would be helpful.

wonderfulShrineMaidenOfParadise, 2021-10-07:
It's possible to check this in stock bootloader.
[image]

Or you can get ``androidboot.hardware.ddr=elpida`` by running this in lk2nd:
```
fastboot oem lk_log && fastboot get_staged /dev/stdout # To stdout
fastboot oem lk_log && fastboot get_staged lk_log.txt # To file lk_log.txt
```

minlexx, 2021-10-14:
I can't confirm. I just built lk2nd from current master and it just works fine.

I have model D821(H) 32 Gb:

[image]

and current lk2nd works.

[image]

lk_log: [lk_log_hh.txt](https://github.com/msm8916-mainline/lk2nd/files/7346545/lk_log_hh.txt)

Justinzobel, 2021-12-05:
> I can't confirm. I just built lk2nd from current master and it just works fine.
> 
> I have model D821(H) 32 Gb:
> 
> [image]
> 
> and current lk2nd works.
> 
> [image]
> 
> lk_log: [lk_log_hh.txt](https://github.com/msm8916-mainline/lk2nd/files/7346545/lk_log_hh.txt)
> 
> I haven't tried to boot TWRP yet, but it boots pmOS just fine.

So if I have the H I should be able to install the lk2nd for a better experience?

wonderfulShrineMaidenOfParadise, 2021-12-06:
@Justinzobel lk2nd is a solution to [pmaports#1227](https://gitlab.com/postmarketOS/pmaports/-/issues/1227), to make sure hammerhead boot correctly and avoid confusion. But it's currently solved with another solution.

Icenowy, 2022-03-13:
0.11.0 works my on D820(H).

stephan-gh, 2022-03-13:
@wonderfulShrineMaidenOfParadise Given the successful test reports, can we close this issue?

wonderfulShrineMaidenOfParadise, 2022-03-13:
lk2nd for hammerhead seems to be ready. Close.

closed 2022-03-13 by wonderfulShrineMaidenOfParadise