msm8916-mainline/lk2nd #152: [SM-G530H] does not appear on fastboot devices

alfr3dosv, 2022-02-28:
I flashed lk2nd-msm8916 using heimdall, I can boot into lk2nd, it shows that it is in fastboot mode, but so far I haven't been able to detect it on windows or linux using `fastboot devices`

I guess it's because the only ones supported in this family are Samsung Galaxy Grand Prime - SM-G530W, SM-G530Y

TravMurav, 2022-02-28:
Hi, for this device lk2nd has to enable usb properly which it only can do if it can detect the device. You have to add something like this:

https://github.com/msm8916-mainline/lk2nd/blob/master/dts/msm8916/msm8916-samsung-r02.dts#L25-L34

for your variant, then lk2nd should work correctly. I believe it should already show the "DTB" line that would hint which dts file to edit.

wonderfulShrineMaidenOfParadise, 2022-03-02:
PR #151 is on the way.

alfr3dosv, 2022-03-02:
Hi thanks for your work, looking forward to test it

PR #151 (2022-03-21): dts: msm8916: Add Samsung Galaxy Grand Prime (SM-G530H)
Closes #152
closed 2022-03-21 by TravMurav