msm8916-mainline/lk2nd #182: [SM-T377A] Heimdall will not install lk2nd because Secure Download is enabled. Is there any workaround?

TheoCGaming, 2022-06-30:
I want to install PostmarketOS, but I can't install lk2nd on my tablet. Does anyone have a similar experience or a workaround or program?

Note: My main OS on my PC is Arch Linux with Flatpak, Pacman, and Git as package managers.

TheoCGaming, 2022-06-30:
(although git isn't really a package manager)

stephan-gh, 2022-07-01:
Are you sure that your tablet actually allows flashing custom software? This sounds a bit like a carrier-locked device that is bootloader-locked.

wonderfulShrineMaidenOfParadise, 2022-07-08:
Not an MSM8916 variant as far as I know.
Reference: https://desktop.firmware.mobi and search for `gtes`

TheoCGaming, 2022-07-08:
>carrier-locked device

It is carrier-locked, yes, but it's already been paid for and AT&T won't give the unlock code for such an outdated device anymore, even if only because the account it was on doesn't exist anymore.

TheoCGaming, 2022-07-08:
I do wonder if there's a bypass for the unlock code... that is legal if you've already paid for the device right?

wonderfulShrineMaidenOfParadise, 2022-07-09:
@TheoCGaming SM-T377A is not MSM8916, so msm8916-mainline/lk2nd doesn't support it, neither does msm8916-mainline/linux. Thus we cannot provide too much support for it. Good luck.
https://wiki.postmarketos.org/wiki/Exynos_mainline_porting

closed 2023-02-25 by stephan-gh