msm8916-mainline/lk2nd #187: [SM-A300FU] lk2nd does not start

gwilhermig, 2022-07-29:
Hello.
I tried to install postmaketOS in my phone (Samsung Galaxy A3 2015 SM-A300FU).
I have flashed lk2nd in my device with heimdall flash --BOOT lk2nd-msm8916.img.
Here is the end of the message that appears on my computer:
"Uploading BOOT
100%
BOOT upload successful
Ending session...
Rebooting device...
Releasing device interface..."
My phone restarted on a page with the message: "Set Warranty Bit : kernel" and nothing else happens. When i try to connect it on my computer, i see the message "ERROR : Failed to detected compatible download-mode device".
When i press on vol down + home + power, i see the screen in odin mode and i can see the device from lsusb.
When i press on vol down + power, i see the screen with "Set Warranty Bit: kernel" and i can't see the device from lsusb.

jvan1234, 2022-07-31:
I'm having a similar issue. Quick question: does your phone also have red text that displays "KERNEL IS NOT SEANDROID ENFORCING"? that could be the potential issue, but I'm a bit newbie, so I'm not sure. Does anyone know why this is happening to us. (Mine's a Samsung Galaxy S4)

stephan-gh, 2022-07-31:
> (Mine's a Samsung Galaxy S4)

We don't have support for any Samsung Galaxy S4 in lk2nd at the moment?

jvan1234, 2022-08-03:
Thank you. I checked my phone and it was not one that was supported, sorry for the false alarm...

gwilhermig, 2022-08-08:
> I'm having a similar issue. Quick question: does your phone also have red text that displays "KERNEL IS NOT SEANDROID ENFORCING"? that could be the potential issue, but I'm a bit newbie, so I'm not sure. Does anyone know why this is happening to us. (Mine's a Samsung Galaxy S4)

I don't have red text that displays "KERNEL IS NOT SEANDROID ENFORCING".

stephan-gh, 2023-02-25:
Please join the chat at [#msm8916:postmarketos.org](https://matrix.to/#/#msm8916:postmarketos.org) (or `#msm8916-mainline` on irc.oftc.net) for device support. Hard to help here in GitHub issues.

closed 2023-02-25 by stephan-gh