msm8916-mainline/lk2nd #189: booting with sdcard inserted "fails"

craigcomstock, 2022-08-14:
I was trying to setup a dual-boot situation with two surnia's (one xt1526, one xt1527). Both seem to exhibit the same behavior.

What I did:
- installed lineageos and confirmed OK
- flashed lk2nd 0.12.0 to boot
- flashed lineageos boot.img from their zip via lk2nd
- confirmed boot to lineageos works through lk2nd
- inserted sdcard created by pmbootstrap
- restarted

I would expect that postmarketos would start, it doesn't.
I would expect that volume down button would get me into lk2nd, it doesn't.

I just get my "unlocked bootloader" screen and seemingly "stuck".

I didn't see anyway that lk_log would help since I can't get to lk2nd when the sdcard is in.

I checked lk_log when sdcard is not in and I see these lines. Do they indicate just that there is no sdcard or that there is some problem with the sdcard reader?

```
[150] found GPT
[160] initialising mmc_slot =2
[160]  Initializing MMC host data structure and clock!
[170] Error: Command timeout error
[170] Failure getting OCR response from MMC Card
[170] MMC card failed to respond, try for SD card
[180] Error: Command timeout error
[180] The response for CMD8 does not match the supplied value
[190] Failed to initialize SD card
[190] Failed detecting MMC/SDC @ slot2
[190] sdcard init failed![200] qhypstub image lacks magic, not loading
[200] pm8x41_get_is_cold_boot: cold boot
[200] calling apps_init()
[210] Display Init: Start
```

TipzRickyCheung, 2022-08-15:
Maybe we need to kick the SD card init stuff, since lk supports SD boot too.

TravMurav, 2022-08-15:
For at least one person so far, a specific sdcard has caused the lk to hang/crash on boot. Other cards did work fine for them. If you always used the same card, maybe try some other one.

Sadly I don't have any such dubious cards so it's very hard to debug.

The log lines do just mean "we failed to talk to the empty slot 2"

TipzRickyCheung, 2022-08-15:
> For at least one person so far, a specific sdcard has caused the lk to hang/crash on boot. Other cards did work fine for them. If you always used the same card, maybe try some other one.
>
> Sadly I [...]

I have a few SD cards here, I could try that out. Maybe it is a partition format issue (MBR and GPT), not sure.

craigcomstock, 2022-08-23:
I think closing this issue is probably OK. Not many are using the feature and I certainly got it working with a different card. This ticket can be found by others. :+1:

closed 2022-08-23 by craigcomstock