msm8916-mainline/lk2nd #196: Question: Is the S-Pen variants of samsung devices supported, e.g., SM-P550?

oliverlew, 2022-09-10:
Reading [PostmarketOS's wiki](https://wiki.postmarketos.org/wiki/Samsung_Galaxy_Tab_A_9.7_2015_(samsung-gt510)) and wondering if s-pen variant like sm-p550 is supported just like sm-t550? I guess yes, but want to make sure.

stephan-gh, 2022-09-10:
They should be 99% the same except for the pen. Not much testing has been done on them so far so if you're interested to run postmarketOS I suggest that you join the [postmarketOS mainline chat](https://wiki.postmarketos.org/wiki/Matrix_and_IRC) and ask for testing instructions there.

oliverlew, 2022-09-10:
Thanks for the reply. As for the issue, I think I got enough information, that is those are similar enough for us to assume they are all supported before actually testing them.

closed 2022-09-10 by oliverlew