msm8916-mainline/lk2nd #197: lk2nd 0.11.0 broke support for htc-m8qlul

Newbytee, 2022-09-17:
Some change between lk2nd 0.10.0 and 0.11.0 broke htc-m8qlul. I haven't bisected this yet, but I'm making this for tracking purposes.

Newbytee, 2022-09-19:
I got this:

```
5f03ceed84a95be6283343c0fbf0ec813557c9b0 is the first bad commit
commit 5f03ceed84a95be6283343c0fbf0ec813557c9b0
Author: Minecrell <minecrell@minecrell.net>
Date:   Thu Oct 28 17:22:40 2021 +0200

    makefile: Drop -fno-builtin
    
    This allows gcc to inline standard calls like strlen("constant string").

 include/stdio.h   | 1 +
 lib/libc/printf.c | 6 ++++++
 makefile          | 2 +-
 3 files changed, 8 insertions(+), 1 deletion(-)
```

I can also confirm that its parent, 9d9c3abca539c6ef3be625e61ec489fbef490d3c, kind of works. The phone repeatedly vibrates on the HTC splash, but if I hold down the power and volume down buttons down for a moment it somehow makes it to a functional lk2nd.

However, reverting 5f03ceed84a95be6283343c0fbf0ec813557c9b0 on top of 0.11.0 does not work. The phone just goes to its own fastboot if I do that, even with the non-m8qlul dtbs removed from the build.

Newbytee, 2022-09-19:
Honestly I'm not sure what to make of this. Previously the device went into EDL mode when I built 5f03ceed84a95be6283343c0fbf0ec813557c9b0, but now it goes into fastboot mode with that commit.

stephan-gh, 2023-01-11:
22:27 <minecrell[m]> "makefile: Drop -fno-builtin" is broken with some gcc versions until https://github.com/msm8916-mainline/lk2nd/commit/1b36977eb809e36c3d6439f22bf9c95cbf824ced
22:27 <minecrell[m]> so that may be a false positive