msm8916-mainline/lk2nd #199: Not recognising SM-T350

oliverlew, 2022-10-03:
The device is Samsung's Galaxy Tab A 8.0, SM-T350. It seems to be supported by https://github.com/msm8916-mainline/lk2nd/pull/186.

Please tell me what went wrong here:

First I cloned this repo and build with the command
```
make TOOLCHAIN_PREFIX=arm-none-eabi- lk2nd-msm8916
```
No errors, then I flashed the image with
```
fastboot flash lk2nd build-lk2nd-msm8916/lk2nd.img
```
When boot into lk2nd, the device is not recognized. It shows:
```
... ...
MODEL - Unknown (FIXME!)
... ...
DTB - <268 0cx01ff01 0 0x4>
... ...
```

The dtb info is different from that PR. But I have no idea what that implies.

oliverlew, 2022-10-03:
There is no `BOOTLOADER` in lk2nd screen, but here is a photo of it so you can see all the info it displays:

[image]

As for the Android version, I installed LineageOS 18.1 on it. But because I am not the original owner of the device, I don't know the stock ROM version it had before.

So if it had Android 5 but not 6 before installation of LOS, could flashing the last version of firmware now help with the issue?

stephan-gh, 2022-10-03:
Can you post the log file using `fastboot oem lk_log` `fastboot get_staged log.txt` then?

oliverlew, 2022-10-04:
Here is the log file
[log.txt](https://github.com/msm8916-mainline/lk2nd/files/9702434/log.txt)

stephan-gh, 2022-10-04:
`T350XXU1AOH2` seems to be indeed an old Android 5 firmware. It's up to you: for lk2nd you could easily work around this issue but if you also want to use the mainline Linux kernel (postmarketOS port) I would recommend that you update to a recent Android 7 build. There are minor firmware differences (such as this one) that may or may not cause trouble.

Feel free to [join the chat](https://wiki.postmarketos.org/wiki/Matrix_and_IRC) (main or mainline channel) if you want help with updating the firmware on your device.

oliverlew, 2022-10-05:
I just updated firmware files from `T350XXU1CQJ5`, and now lk2nd really recognized the device!

So as far as this issue, I think it is solved. Thanks again!

closed 2022-10-05 by oliverlew