msm8916-mainline/lk2nd #200: Can't flash

MichalkapDeveloper, 2022-10-06:
I cant flash to Moto G4 Play harpia.
Terminal output:
`(bootloader) has-slot:boot: not found
(bootloader) is-logical:boot: not found
Sending 'boot' (494 KB)                            OKAY [  0.042s]
Writing 'boot'                                     (bootloader) Image not signed or corrupt

OKAY [  0.052s]
Finished. Total time: 0.153s`

MichalkapDeveloper, 2022-10-06:
: boot : always turns into :boot:

stephan-gh, 2022-10-06:
Try `fastboot boot lk2nd.img` first then do `fastboot flash lk2nd lk2nd.img`.

MichalkapDeveloper, 2022-10-06:
Does not work

stephan-gh, 2022-10-06:
Why?

ph1-xyz, 2022-12-13:
Similar problem here, but with motorola-osprey

ph1-xyz, 2022-12-13:
> Why do you select recovery if you want to boot into lk2nd?

I may be doing the wrong thing, but either rebooting seemed to bring me again to the Fastboot menu :-/

stephan-gh, 2023-02-25:
Please join the chat at [#msm8916:postmarketos.org](https://matrix.to/#/#msm8916:postmarketos.org) (or `#msm8916-mainline` on irc.oftc.net) for device support. Hard to help here in GitHub issues.

closed 2023-02-25 by stephan-gh