msm8916-mainline/lk2nd #204: Lenovo A6010 won't boot from ln2nd to recovery.

mihas1984, 2022-10-24:
I am installing PostmarketOS on a Lenovo A6010 specially bought for this purpose. The wiki https://wiki.postmarketos.org/wiki/Lenovo_A6010_(lenovo-a6010) says it has a charging problem: it's recommended to go back into TWRP to charge, because native charger doesn't work. I did everything according to the instructions, successfully flashed lk2nd, installed the system on userdata, everything is fine. But. I can't go into recovery now from lk2nd. 

Wiki says:
Fastboot: Power on the device with both Power and Volume Down buttons.
- it works
Recovery: Power on the device with Volume Down, Volume Up, and Power buttons. 
- it works
lk2nd Fastboot: Power on the device. After it vibrates, hold Volume Down. 
- it works
lk2nd Recovery: Power on the device. After it vibrates, hold Volume Up.
- does not work, still enters lk2nd. 

If you select “Recovery mode” in lk2nd itself, you return to lk2nd. 

If you flash the recovery, it flashes successfully. But it still doesn't go in. If you try to make fastboot boot recovery.img from lk2nd, it says 

C:\adb_tools>fastboot boot recovery.img
Sending 'boot.img' (24524 KB)                      OKAY [  0.766s]
Booting                                            FAILED (remote: 'dtb not found')
fastboot: error: Command failed

In native bootloader this command works fine. 

It is possible to enter recovery from native bootloader (to boot into native fastboot, then make fastboot boot recovery.img from native fastboot or using Volume Down, Volume Up, and Power buttons). After that, TWRP is successfully launched. So the problem is with lk2nd. If native bootloader successfully loads recovery, then lk2nd for some reason refuses to do it. Although it loads the PostmarketOS system itself, so I conclude that it is installed normally and copes with the main function.

I use last version of lk2nd, 0.13

stephan-gh, 2022-10-24:
Please post the

```
fastboot oem lk_log
fastboot get_staged log.txt
```

mihas1984, 2022-10-24:
[log.txt](https://github.com/msm8916-mainline/lk2nd/files/9853856/log.txt)

mihas1984, 2022-10-24:
First log was after usual boot. This log is after trying to reboot to recovery (it differs)/ 
[log.txt](https://github.com/msm8916-mainline/lk2nd/files/9853868/log.txt)

stephan-gh, 2022-10-24:
Can you post a link to the TWRP image?

mihas1984, 2022-10-24:
https://drive.google.com/file/d/1F-mbJnM96RgFvc0mH30lw5SyG0SkNtCS/view?usp=sharing

mihas1984, 2022-10-26:
I continue to experiment with loading recovery. I tried to use the TWRP version from a similar Lenovo A6000 phone (link https://drive.google.com/file/d/1aGOF1SA-PthKRriIIiU5yWOevuQ7NR6n/view?usp=sharing). At the same time, the result is completely opposite: the native A6010 bootloader refuses to load it,

C:\adb_tools>fastboot boot recovery_6000.img
Sending 'boot.img' (22982 KB) OKAY [ 0.723s]
Booting FAILED (remote: 'dtb not found')
fastboot: error: Command failed

While lk2nd loaded it successfully. But since the A6000 phone has a different charge controller, I returned back to my native recovery from the A6010 and issue is still open.

For some reason, two bootloaders successfully load different recovery: if the main one loads from A6010, then lk2nd refuses; if the main one refuses to load from the A6000, then lk2nd successfully loads.

mihas1984, 2022-10-27:
[image]
I have added this lines in dts/msm8916/msm8916-qrd9-v1.dts and now it successfuly boots to recovery from both bootloaders, native and lk2nd. Thank you!