msm8916-mainline/lk2nd #209: Galaxy A3 with product name SM-A3009 is not supported.

water-chika, 2022-11-08:
When I see that a Samsung Galaxy A3 phone could run postmarketOS, I buy one. But when I try to run lk2nd, I find that my Galaxy A3 is SM-A3008 and that one is not supported by lk2nd. I have tryed some methods such as modifying dts... But I do not known which downstream is about SM-A3009. Is there are some quick method to add support for SM-A3009. Or some help that could let me known how to add support for a variant of Samsung Galaxy A3.
Thanks for reading.

stephan-gh, 2022-11-08:
What happens if you flash lk2nd? Does it boot?

water-chika, 2022-11-08:
> What happens if you flash lk2nd? Does it boot?

It shows red text "Could not do normal boot." in Samsung Download mode

stephan-gh, 2022-11-08:
Are you sure the device actually allows flashing custom software? A3009 seems to be a carrier device (China Telecom) and I'm not sure what the rules for those are. Is there something like TWRP for this device that you could try?

water-chika, 2022-11-08:
First 

> > What happens if you flash lk2nd? Does it boot?
> 
> It shows red text "Could not do normal boot." in Samsung Download mode

It will flash success, and reboot. But after reboot it will re-enter Samsung Download mode and show that text in above other text.

water-chika, 2022-11-08:
Yes, I tried TWRP. And that could run recovery mode.

stephan-gh, 2022-11-08:
Can you post a link to the TWRP image that works correctly?

water-chika, 2022-11-08:
https://dl.twrp.me/a3xelte

stephan-gh, 2022-11-08:
a3xelte is for the A3 (2016) though and not the A3 (2015). Are you sure you actually have a A3009 with Qualcomm Snapdragon 410/MSM8916?

water-chika, 2022-11-08:
I plan to install LineageOS, but there is not a rom in official website. And build a rom need a lot of time and a lot of disk space.

water-chika, 2022-11-08:
Yes, it shows "product name: SM-A3009"

stephan-gh, 2022-11-08:
https://dl.twrp.me/a3xelte shouldn't work on SM-A3009 though...

water-chika, 2022-11-08:
After I flash it into RECOVERY partition, it could go into recovery mode by pressing Volumn Up + Home + Power.

water-chika, 2022-11-08:
Let me try flashing some other data into this partion. And I can not go into recovery mode now.
Try flashing twrp, it shows "Failed to confirm end of file transfer sequence!" in my terminal that run "heimdall flash --RECOVERY twrp.img".
Maybe there are some problems in this phone. I will try some other method and I will tell you if  this phone could run properly.
Thanks for your help!

water-chika, 2022-11-08:
I tryied run "heimdall flash --RECOVERY twrp.img --no-reboot", and it show "binary size is too large : recovery".
It is my problem that I do not flash it into RECOVERY.

stephan-gh, 2022-11-08:
Can you try this one? (please unpack it first) [lk2nd-a3.zip](https://github.com/msm8916-mainline/lk2nd/files/9963614/lk2nd-a3.zip)

water-chika, 2022-11-09:
It shows yellow texts "Set Warranty Bit : kernel"

stephan-gh, 2022-11-09:
And then? Nothing happens?

water-chika, 2022-11-09:
Nothing happened. Just show samsung logo or show charging picture. The yellow texts show above logo or picture.

wonderfulShrineMaidenOfParadise, 2022-11-14:
Please join the matrix chat, [#mainline:postmarketos.org](https://matrix.to/#/#mainline:postmarketos.org) for porting.

wonderfulShrineMaidenOfParadise, 2023-01-15:
We have new matrix room now. Please join [#msm8916:postmarketos.org](https://matrix.to/#/#msm8916:postmarketos.org) if you still need help with porting.

closed 2023-02-25 by stephan-gh