msm8916-mainline/lk2nd #216: About Samsung SM-A7000

Oopas, 2022-11-14:
I see that this project has support for Samsung Galaxy A7 (2015) - SM-A700YD

The SM-A7000 is its Chinese version (the same processor MSM8939)

This is the official open source kernel source code of this model (4.4-6.0): [Kernel_Source_Code](https://opensource.samsung.com/uploadSearch?searchValue=A7000)

This is the twrp image I normally use: [SM-A7000_twrp](https://anonfiles.com/K3T2tbHfy2/SM-A7000_twrp_img) 

Can you complete it ? Thank you !

stephan-gh, 2022-11-14:
cc @wonderfulShrineMaidenOfParadise

wonderfulShrineMaidenOfParadise, 2022-11-14:
Please join the matrix chat, [#mainline:postmarketos.org](https://matrix.to/#/#mainline:postmarketos.org) for porting.

wonderfulShrineMaidenOfParadise, 2022-11-14:
Or [#msm8939:matrix.org](https://matrix.to/#/#msm8939:matrix.org) since it's msm8939.

wonderfulShrineMaidenOfParadise, 2023-01-15:
We have new matrix room now. Please join [#msm8916:postmarketos.org](https://matrix.to/#/#msm8916:postmarketos.org) if you still need help with porting.

closed 2023-02-25 by stephan-gh