msm8916-mainline/lk2nd #241: `fastboot flash partition` doesn't work on lk1st

iovxw, 2023-03-10:
Is this expected behavior?
```
$ fastboot getvar product
product: LK1ST_MSM8916
Finished. Total time: 0.002s

$ fastboot flash partition ./gpt_both0.bin
Sending 'partition' (33 KB)                        OKAY [  0.005s]
Writing 'partition'                                FAILED (remote: 'partition table doesn't exist')
fastboot: error: Command failed
```

https://github.com/msm8916-mainline/lk2nd/blob/627d2198269c6826d1056d863ae867199ab04257/app/aboot/aboot.c#L2696-L2706

stephan-gh, 2023-03-10:
Currently this is disabled on purpose for safety reasons, since users might otherwise easily corrupt the entire storage by flashing a wrong partition table.

iovxw, 2023-03-11:
Perhaps we can add a compile time option to enable it? Or a new oem command (with appropriate warnings) to enable it at runtime?

185264646, 2023-03-15:
> Perhaps we can add a compile time option to enable it? Or a new oem command (with appropriate warnings) to enable it at runtime?

I suggest doing this from linux, the tools are more stable and reliable. Refer to [jumpdrive](https://github.com/dreemurrs-embedded/Jumpdrive) and sgdisk scripts.