msm8916-mainline/lk2nd #263: Not detecting my device model (J5 2016 - J510F)

Ygor6889, 2023-06-25:
I'm trying to flash postmarketOS with Phosh interface using the pmbootstrap flasher method but I keep getting stuck on "Waiting for devices" and flashing using the TWRP recovery zip error I got "Error 1"

Both this errors are related with "MODEL - Unknown (FIXME!)" showing on the lk2nd fastboot screen, I tried with two J5 2016 models (J510MN and J510F) and I got the same results, looks like lk2nd isn't detecting my phone model

Is that a bug or I need to do something else?

TravMurav, 2023-06-25:
Hi, this device needs lk2nd to know about the MUIC chip inside to enable USB. This means lk2nd must properly detect the device.

For J510MN you would likely need to add the variant to the lk2nd control DT, see i.e. https://github.com/msm8916-mainline/lk2nd/commit/736e1b4ba2ca732bb7193e9e33555ef8dc194d9a

J510F is [already added](https://github.com/msm8916-mainline/lk2nd/blob/master/dts/msm8916/msm8916-samsung-r04.dts#L13-L22) so it should have worked, however samsung bootloaders are pretty messy so you may need to i.e. update to the latest firmware before it starts to boot lk2nd correctly; alternatively you may have to add another DT node for that in a different file. You can see what file is selected using the DTB field on the screen. (i.e. it's samsung-r04 as there is 0x04 on the screenshot)

wonderfulShrineMaidenOfParadise, 2023-06-25:
Please join [#msm8916:postmarketos.org](https://app.element.io/#/room/#msm8916:postmarketos.org) for faster helps.

Ygor6889, 2023-06-25:
Hey thank you for the response, but how can I add my J510MN variant to the lk2nd control DT? About the other one, I'm gonna install the latest stock ROM

closed 2023-06-25 by Ygor6889