msm8916-mainline/lk2nd #274: Support for MSM8960

AtulBha, 2023-07-06:
Looking at the article here: https://wiki.postmarketos.org/wiki/MSM8960_Mainlining
and looking at DTS files it does not appear that there is support for A700FD yet. 

I verified that by flashing lk2nd but did not have success with fastboot. It is possible to request dts/port/support for the said device

TravMurav, 2023-07-09:
Hi, if you need support for msm8960 (Snapdragon S4), you would need to use WIP experimental-tmp2 branch which has some level support for it.

However, "A700FD" (which I assume is Samsung Galaxy A7) is a snapdragon 615 based device - msm8939, and both the platform, and a different regional variant of this device are supported in the current lk2nd branch, so you should just need to add a dt node for it, like for the other a7.

AtulBha, 2023-07-12:
I will trust your assessment, I was going by what I see when I plug in the device. I really am not sure how to generate the metadata file, where do I find all the relevant information to create it.