msm8916-mainline/lk2nd #278: Galaxy A5 SM-A5000 MODEL - Unknown (FIXME!)

cyberstudio10, 2023-07-24:
I was able to flash lk2nd with heimdall, thanks! It boots up and say MODEL - Unknown (FIXME!). I had to disable driver signing and add ```%SingleBootLoaderInterface% = USB_Install, USB\VID_18D1&PID_D00D``` to Google's usb driver INF file before Windows would recognize lk2nd as a fastboot device.

This is all new to me... I had zero idea whether everybody has to disable driver signing and add PID=D00D like this, before fastboot will recognize the lk2nd device. Or is it only because my MODEL is Unknown that I have to install an unsigned driver? I had no idea how the model detection works. My model is SM-A5000. Let me know if I need to retrieve more information to identify the model of my phone. Thank you.

TravMurav, 2023-07-25:
Hi, we generally don't expect the user to use Windows so it's on the user to know what to do, lk2nd should use generic fastboot usb ids though.

As per A5000, the configuration was only recently added (in https://github.com/msm8916-mainline/lk2nd/commit/67d801a85eb93fa3817ed6f7235b3d057ed7fafe) and there was no new lk2nd release yet to include it. You should be able to build lk2nd from source to see if the device model is detected properly.

vicenteicc2008, 2023-07-29:
I am having same problem on my s5 mini

wonderfulShrineMaidenOfParadise, 2023-08-04:
See https://github.com/msm8916-mainline/lk2nd#building for how to build.
Consider joining [#msm8916:postmarketos.org](https://app.element.io/#/room/#msm8916:postmarketos.org) for helps.

wonderfulShrineMaidenOfParadise, 2024-07-04:
Looks similar to #385