msm8916-mainline/lk2nd #281: S5 Mini Duos USB not working

vicenteicc2008, 2023-08-02:
my computer not detecting my s5 mini with lk2nd

vicenteicc2008, 2023-08-03:
is msm8226

wonderfulShrineMaidenOfParadise, 2023-08-04:
You might need to add a node, `samsung,muic-reset`.
https://github.com/msm8916-mainline/lk2nd/blob/master/dts/msm8916/msm8916-samsung-cprime.dtsi#L1-L4

Check your MUIC type and GPIO pins in downstream dts and change the values accordingly.

vicenteicc2008, 2023-08-04:
muic-reset is not in the dts

wonderfulShrineMaidenOfParadise, 2023-08-04:
Uh, so, add it then?

vicenteicc2008, 2023-08-04:
but in the downstream it is not

wonderfulShrineMaidenOfParadise, 2023-08-04:
> Check your MUIC type and GPIO pins in downstream dts

No. You don't look for it in downstream. You are looking for `MUIC type and GPIO pins`

vicenteicc2008, 2023-08-04:
but how do i find it?

wonderfulShrineMaidenOfParadise, 2023-08-04:
Ctrl + F and look for `sm5502` or `sm5504`

vicenteicc2008, 2023-08-04:
i found it

vicenteicc2008, 2023-08-04:
How can I know the pins and the address?

wonderfulShrineMaidenOfParadise, 2023-08-04:
Look for keywords, `gpio` and `reg`.

vicenteicc2008, 2023-08-04:
same problem

wonderfulShrineMaidenOfParadise, 2023-08-04:
I am afraid that I don't understand your problem without codes pushed to GitHub.

wonderfulShrineMaidenOfParadise, 2023-08-04:
Consider joining [#lk2nd:postmarketos.org](https://app.element.io/#/room/#lk2nd:postmarketos.org) for helps in chats.