msm8916-mainline/lk2nd #290: dts: msm8916: fortuna3g: The computer does not detect the cell phone in fastboot mode

pachof, 2023-08-28:
Describe your issue
What's the expected behaviour?

the computer detects the cell phone in fastboot mode

What's the current behaviour?

the computer does not detect the cell phone in fastboot mode

How to reproduce your issue?

    Download [lk2nd-msm8916.img](https://github.com/msm8916-mainline/lk2nd/releases/download/0.14.0/lk2nd-msm8916.img).
    
    Flash with hemidall: heimdall flash --BOOT lk2nd.img

    Boot into lk2nd and connect the usb to cellphone
    See the errors.

What device are you using?

samsung-fortuna3g SM-G530H
On what lk2nd version did you encounter the issue?

0.14.0

wonderfulShrineMaidenOfParadise, 2023-08-29:
Please join [#msm8916:postmarketos.org](https://app.element.io/#/room/#msm8916:postmarketos.org) for helps.

wonderfulShrineMaidenOfParadise, 2023-08-31:
Note that your Grand Prime is not supposed to be `SM-G530H` or `fortuna3g`.
Once you have your works done, please consider opening a pull request.

exkc, 2023-09-29:
fastboot not hemidall

wonderfulShrineMaidenOfParadise, 2023-09-29:
> use fastboot in lk2nd not hemidall

Well yes. On Samsung devices you have to flash it with heimdall first. Also lk2nd fastboot is broken on all of Core Prime / Grand Prime series, which requires MUIC reset, and extra patches for dts.