msm8916-mainline/lk2nd #291: Codeaurora links in readme are broken

Newbytee, 2023-09-03:
The readme links to Codeaurora in several instances. As I understand, Codeaurora was permanently shut down.

 - https://github.com/msm8916-mainline/lk2nd/blob/e13be68f4825ee898141dee9374f8aa8374449c6/README.md?plain=1#L35C1-L35C48
 - https://github.com/msm8916-mainline/lk2nd/blob/e13be68f4825ee898141dee9374f8aa8374449c6/README.md?plain=1#L211C13-L211C61

z3ntu, 2023-09-12:
Also in `LICENSE` and `target/msm8916/qhypstub_loader.c` the links should be updated.

The new source location should be https://git.codelinaro.org/clo/la/kernel/lk but I cannot seem to find the referenced tags there, no clue if they just messed up the migration or if I'm looking at the wrong place.

z3ntu, 2023-09-12:
Ah and duplicate of #244