msm8916-mainline/lk2nd #306: Redmi 3(ido) cannot enter the main interface of postmarketOS after being brushed into it

alabohai, 2023-09-29:
After brushing into PostmarketOS, Redmi 3(ido) cannot enter the main interface of PostmarketOS. It keeps displaying the loading interface of PostmarketOS, and then the screen becomes dark, not black, but nothing can be seen. Ubuntu can connect to a Redmi 3(ido)  phone using SSH and install Docker, but the Docker operation failed with a prompt indicating 'Failed to start daemon: Devices cgroup is not mounted'.

wonderfulShrineMaidenOfParadise, 2024-06-12:
There are ido porting things done recently, especially the panel display.
Consider joining [#msm8916-mainline:matrix.org](https://matrix.to/#/#msm8916:postmarketos.org) for faster helps.

orange200213, 2025-04-14:
> There are ido porting things done recently, especially the panel display. Consider joining [#msm8916-mainline:matrix.org](https://matrix.to/#/#msm8916:postmarketos.org) for faster helps.

Hello，I noticed that there is lk2nd support for redmi3 ido but there is no device tree file. I wrote a simple dts file and use an arch rootfs it can boot with backlight. I think it succeeded booting but I can not get more information.Can I get the dts file somewhere？ I am so sorry to disturb you but I would appreciate it if someone can help me.
Thank you！

orange200213, 2025-04-14:
I am a new bee and with the help of chatgpt I wrote these. ``
```
/*
 * Xiaomi Redmi 3 (ido)
 */

/dts-v1/;

#include "msm8939.dtsi"
#include "msm8939-pm8916.dtsi"
#include <dt-bindings/arm/qcom,ids.h>
#include <dt-bindings/gpio/gpio.h>

/ {
	model = "Xiaomi Redmi 3";
	compatible = "xiaomi,ido", "qcom,msm8939";
	chassis-type = "handset";

	qcom,board-id = <QCOM_BOARD_ID_MTP 0>;
	qcom,msm-id = <QCOM_ID_MSM8939 0>, <QCOM_ID_MSM8939 0x30000>;

	aliases {
		serial0 = &blsp_uart2;
		mmc0 = &sdhc_1;
		mmc1 = &sdhc_2;
	};

	chosen {
		stdout-path = "serial0:115200n8";
	};

	backlight: backlight {
		compatible = "pwm-backlight";
		pwms = <&pm8916_pwm 0 50000>;
		brightness-levels = <0 32 64 128 192 255>;
		default-brightness-level = <4>;
	};

	panel_power: regulator-panel {
		compatible = "regulator-fixed";
		regulator-name = "vdd-lcd";
		regulator-min-microvolt = <2800000>;
		regulator-max-microvolt = <2800000>;
		gpio = <&tlmm 91 GPIO_ACTIVE_HIGH>;
		enable-active-high;
	};
```
};

&mdss {
	status = "okay";
};

&mdss_dsi0 {
	status = "okay";

	panel@0 {
		compatible = "boe,hx8394f";
		reg = <0>;
		reset-gpios = <&tlmm 25 GPIO_ACTIVE_LOW>;
		enable-gpios = <&tlmm 26 GPIO_ACTIVE_HIGH>;
		vdd-supply = <&panel_power>;
		backlight = <&backlight>;
	};
};

&pm8916_pwm {
	status = "okay";
};

&sdhc_1 {
	status = "okay";
};

&sdhc_2 {
	status = "okay";
};

orange200213, 2025-04-14:
I am a new bee and with the help of chatgpt I wrote these.
`/*
 * Xiaomi Redmi 3 (ido)
 */

/dts-v1/;

#include "msm8939.dtsi"
#include "msm8939-pm8916.dtsi"
#include <dt-bindings/arm/qcom,ids.h>
#include <dt-bindings/gpio/gpio.h>

/ {
	model = "Xiaomi Redmi 3";
	compatible = "xiaomi,ido", "qcom,msm8939";
	chassis-type = "handset";

	qcom,board-id = <QCOM_BOARD_ID_MTP 0>;
	qcom,msm-id = <QCOM_ID_MSM8939 0>, <QCOM_ID_MSM8939 0x30000>;

	aliases {
		serial0 = &blsp_uart2;
		mmc0 = &sdhc_1;
		mmc1 = &sdhc_2;
	};

	chosen {
		stdout-path = "serial0:115200n8";
	};

	backlight: backlight {
		compatible = "pwm-backlight";
		pwms = <&pm8916_pwm 0 50000>;
		brightness-levels = <0 32 64 128 192 255>;
		default-brightness-level = <4>;
	};

	panel_power: regulator-panel {
		compatible = "regulator-fixed";
		regulator-name = "vdd-lcd";
		regulator-min-microvolt = <2800000>;
		regulator-max-microvolt = <2800000>;
		gpio = <&tlmm 91 GPIO_ACTIVE_HIGH>;
		enable-active-high;
	};
};

&mdss {
	status = "okay";
};

&mdss_dsi0 {
	status = "okay";

	panel@0 {
		compatible = "boe,hx8394f";
		reg = <0>;
		reset-gpios = <&tlmm 25 GPIO_ACTIVE_LOW>;
		enable-gpios = <&tlmm 26 GPIO_ACTIVE_HIGH>;
		vdd-supply = <&panel_power>;
		backlight = <&backlight>;
	};
};

&pm8916_pwm {
	status = "okay";
};

&sdhc_1 {
	status = "okay";
};

&sdhc_2 {
	status = "okay";
};`

orange200213, 2025-04-14:
I am a new bee and with the help of chatgpt I wrote these.

```

 /*
 * Xiaomi Redmi 3 (ido)
 */

/dts-v1/;

#include "msm8939.dtsi"
#include "msm8939-pm8916.dtsi"
#include <dt-bindings/arm/qcom,ids.h>
#include <dt-bindings/gpio/gpio.h>

/ {
	model = "Xiaomi Redmi 3";
	compatible = "xiaomi,ido", "qcom,msm8939";
	chassis-type = "handset";

	qcom,board-id = <QCOM_BOARD_ID_MTP 0>;
	qcom,msm-id = <QCOM_ID_MSM8939 0>, <QCOM_ID_MSM8939 0x30000>;

	aliases {
		serial0 = &blsp_uart2;
		mmc0 = &sdhc_1;
		mmc1 = &sdhc_2;
	};

	chosen {
		stdout-path = "serial0:115200n8";
	};

	backlight: backlight {
		compatible = "pwm-backlight";
		pwms = <&pm8916_pwm 0 50000>;
		brightness-levels = <0 32 64 128 192 255>;
		default-brightness-level = <4>;
	};

	panel_power: regulator-panel {
		compatible = "regulator-fixed";
		regulator-name = "vdd-lcd";
		regulator-min-microvolt = <2800000>;
		regulator-max-microvolt = <2800000>;
		gpio = <&tlmm 91 GPIO_ACTIVE_HIGH>;
		enable-active-high;
	};
};

&mdss {
	status = "okay";
};

&mdss_dsi0 {
	status = "okay";

	panel@0 {
		compatible = "boe,hx8394f";
		reg = <0>;
		reset-gpios = <&tlmm 25 GPIO_ACTIVE_LOW>;
		enable-gpios = <&tlmm 26 GPIO_ACTIVE_HIGH>;
		vdd-supply = <&panel_power>;
		backlight = <&backlight>;
	};
};

&pm8916_pwm {
	status = "okay";
};

&sdhc_1 {
	status = "okay";
};

&sdhc_2 {
	status = "okay";
};

```