msm8916-mainline/lk2nd #309: Debug(SWD/UART/Trace) over SDC2 enabling

fxsheep, 2023-10-04:
According to the TRM(`lm80-p0436-100_d_snapdragon_410e_apq8016e_tech_reference_manual_revd.pdf`, page 3357), msm8916 supports NIDnT(Narrow Interface for Debug and Test) feature, implemented in a QDSD(Qualcomm Debug over SD?) block inside TLMM. 

Basically, this means that SWD, UART and Trace signals can be routed to microSD slot. On an unfused device, SWD is especially useful for debugging purposes. This is potentionally quite helpful for people who want to contribute msm8916-mainline and relevant low-level projects, but only have a regular OEM device(which normally doesn't have convenient access to JTAG and UART) that has secure boot disabled.

After some trial and error I found the following procedure work for me:
 - 1.Connect a microSD pinout/extender board to the slot(can be made of PCB, or for situation like wt88047 where slot is blocked by the battery, made of [FPC cable](https://www.google.com/url?sa=i&url=http%3A%2F%2Fwww.bplus.com.tw%2FExtenderBoard%2FB18-tw.html&psig=AOvVaw0hxoFh9LUyWpdyJXtnTY-s&ust=1696497877385000&source=images&cd=vfe&opi=89978449&ved=0CBMQjhxqFwoTCJi9j52J3IEDFQAAAAAdAAAAABAI)). 
 - 2.Connect a card into the extender, and remove it later. This step is necessary for Linux/Android kernel on some devices that implement card detect GPIO to enable the SDC2 power domain.
 - 3. Write 0x3c5 to 0x0119e000(`TLMM_QDSD_CONFIG_CTL`) , e.g. `devmem 0x0119e000 32 0x3c5`. (This switches to SWD+UART mode)
 - 4. Connect a SWD adapter to the pinout board, note that SDC2 has either 2.8V or 1.8V voltage levels, and the adapter Vtref has to support them.
 - 5. Try to connect to CoreSight/QDSS DAP via openocd or other host tools.

and I've got Debug ROM table via openocd:
```
> msm8916.dap info 0
AP ID register 0x14770004
	Type is MEM-AP AXI3 or AXI4
MEM-AP BASE 0x00000002
	No ROM table present

> msm8916.dap info 1
AP ID register 0x44770002
	Type is MEM-AP APB2 or APB3
MEM-AP BASE 0x80000000
	ROM table in legacy format
		Component base address 0x80000000
		Peripheral ID 0x00500f0070
		Designer is 0x070, Qualcomm
		Part is 0x070, Unrecognized 
		Component class is 0x1, ROM table
		MEMTYPE system memory not present: dedicated debug bus
	ROMTABLE[0x0] = 0x1003
		Component base address 0x80001000
		Peripheral ID 0x00001f0440
		Designer is 0x070, Qualcomm
		Part is 0x440, Qualcomm QDSS Component v1 (Qualcomm Designed CoreSight Component v1)
		Component class is 0x9, CoreSight component
		Type is 0x04, Debug Control, other
	ROMTABLE[0x4] = 0x2003
		Component base address 0x80002000
		Peripheral ID 0x04001bb962
		Designer is 0x23b, ARM Ltd
		Part is 0x962, CoreSight STM (System Trace Macrocell)
		Component class is 0x9, CoreSight component
		Type is 0x63, Trace Source, Software
	ROMTABLE[0x8] = 0x10003
		Component base address 0x80010000
		Peripheral ID 0x04004bb906
		Designer is 0x23b, ARM Ltd
		Part is 0x906, CoreSight CTI (Cross Trigger)
		Component class is 0x9, CoreSight component
		Type is 0x14, Debug Control, Trigger Matrix
	ROMTABLE[0xc] = 0x11003
		Component base address 0x80011000
		Peripheral ID 0x04004bb906
		Designer is 0x23b, ARM Ltd
		Part is 0x906, CoreSight CTI (Cross Trigger)
		Component class is 0x9, CoreSight component
		Type is 0x14, Debug Control, Trigger Matrix
	ROMTABLE[0x10] = 0x12003
		Component base address 0x80012000
		Peripheral ID 0x04004bb906
		Designer is 0x23b, ARM Ltd
		Part is 0x906, CoreSight CTI (Cross Trigger)
		Component class is 0x9, CoreSight component
		Type is 0x14, Debug Control, Trigger Matrix
	ROMTABLE[0x14] = 0x13003
		Component base address 0x80013000
		Peripheral ID 0x04004bb906
		Designer is 0x23b, ARM Ltd
		Part is 0x906, CoreSight CTI (Cross Trigger)
[... 413 lines cut ...]
		Type is 0x13, Trace Source, Processor
	ROMTABLE[0x18] = 0x0
		End of ROM table

> msm8916.dap info 4
AP ID register 0x00000000
No AP found at this ap 0x4
```

QDSS needs some weird options to work, but with current mainline OpenOCD support, it's possible to debug:
 - APSS(Cortex-A53, debug registers directly exposed via MEM-AP)
 - RPM(Cortex-M3, similar to above)

While it's not yet possible to debug:
 - WCNSS(ARM926EJ-S, requires direct JTAG instructions, so it's behind JTAG-AP, not yet supported by OpenOCD)
 - VENUS(ARM926EJ-S, same as above)
 - Modem+Audio DSP(QDSP6v5, same as above + debug spec not public)
 - Modem-??? (Cortex-M3, ???????)
 - maybe more cores

This may deserve a project on its own(adapters & devices list, openocd configurations), but it would also be great to have an 'enabler' implemented in lk2nd as a oneliner `oem` command. (Requires TZ to have no memory protection enabled, e.g. using ATF)

The usage of QDSD/NIDnT registers are documented in downstream kernel:
https://github.com/LineageOS/android_kernel_wingtech_msm8916/blob/beb5692ed25808aebce7b57e318cf8bd33e645d8/drivers/coresight/coresight-nidnt.h#L21
https://github.com/LineageOS/android_kernel_wingtech_msm8916/blob/beb5692ed25808aebce7b57e318cf8bd33e645d8/drivers/coresight/coresight-nidnt.c#L155

stephan-gh, 2023-10-05:
Nice work figuring this out!

> This may deserve a project on its own(adapters & devices list, openocd configurations), but it would also be great to have an 'enabler' implemented in lk2nd as a oneliner oem command.

I guess you can use `fastboot oem writel 0x0119e000 0x3c5` as one liner? :D

It would be great to document this properly somewhere, also together with instructions for OpenOCD etc.

fxsheep, 2023-10-05:
Yeah, the actual part of toggling QDSD registers can be done easily, but since QDSD documentation in the TRM is rather scarce, how it actually works is still not clear. I've got varying successibility on different devices. My assumption is that SDC2 power has to be enabled(which means this can be device specific), along with some clocks. Meanwhile, a look at CoreSight driver in downstream kernel seems to imply that NIDnT actually support automatic switching(maybe with a special GPIO like card detect?), but again there's no clear document about this.

I'm going to document this once I got stuff clearer.

stephan-gh, 2023-10-05:
> My assumption is that SDC2 power has to be enabled(which means this can be device specific)

FWIW: The regulators for SDC2 are typically powered up by default during boot on MSM8916. We have SD card support in lk2nd without turning on any regulators. The regulators may turn off any time though once Linux is started (if no SD card is inserted).

> along with some clocks

Not sure about the boot up state for those, they are probably off by default. But it seems a bit strange that the debug logic would receive the clock intended for the SDC2 SDHCI controller. Looks like downstream references the regulators typically used for SDC2 and the QDSS clocks managed by RPM: https://github.com/msm8916-mainline/linux-downstream/blob/b20608408caff817ec874f325127b07609fbaeb8/arch/arm/boot/dts/qcom/msm8916-coresight.dtsi#L55-L64

There is also pinctrl (mainly configuring pull up or pull down on the SDC pins). Downstream suggests this should be set depending on the selected mode: https://github.com/msm8916-mainline/linux-downstream/blob/b20608408caff817ec874f325127b07609fbaeb8/arch/arm/boot/dts/qcom/msm8916-qrd.dtsi#L184-L205