msm8916-mainline/lk2nd #310: Support request

shashankx86, 2023-10-04:
Hi,

Can you add support for MDM9207
Thank you

stephan-gh, 2023-10-05:
The experimental-tmp3 branch should have support for MDM9207. If you need specific help, please join the chat (e.g. [#lk2nd:postmarketos.org on Matrix](https://matrix.to/#/#lk2nd:postmarketos.org)).

shashankx86, 2023-10-05:
I am not asking for exact ETA, 
how much estimated time it may take come at main
and is it usable?

stephan-gh, 2023-10-05:
It's quite usable already.

shashankx86, 2023-10-05:
experimental-tmp3, right?

shashankx86, 2023-10-05:
@stephan-gh checked
MDM9207 not there, i think you are confused with MDM9607

shashankx86, 2023-10-05:
@stephan-gh 
one more thing 
as build guide says 
```
$ make TOOLCHAIN_PREFIX=arm-none-eabi- lk2nd-msmXXXX
```
but my device doesn't have any cpu, it just uses MDM9207
and in boot logs it show CPU as 
```
<6>[    0.000000] CPU: ARMv7 Processor [410fc075] revision 5 (ARMv7), cr=10c53c7d
```

full boot logs here
https://github.com/ExtendedFi/JMR540_proc_Dump/blob/main/kmsg

stephan-gh, 2023-10-05:
`make TOOLCHAIN_PREFIX=arm-none-eabi- lk2nd-mdm9607`

xxhg, 2023-12-31:
Can it support SM8250?

closed 2023-12-31 by shashankx86

xxhg, 2023-12-31:
> `make TOOLCHAIN_PREFIX=arm-none-eabi- lk2nd-mdm9607`

Can it support SM8250?