msm8916-mainline/lk2nd #318: On Galaxy J5 2016 (J510H) fastboot doesnt work

Ilya114, 2024-01-06:
hi i trying install postmarketos on Galaxy J5 2016 (J510H) and cant flash rootfs due a PC doesnt detect phone as fastboot device when in lk2nd mode, there is a fix?

TravMurav, 2024-01-07:
if your device shows "model = unknown" on the screen, you will need to add it to the lk2nd's device database.

You would need to add something like this to a correct `dts/msm8916/msm8216-samsung-r<XX>.dts` file:

https://github.com/msm8916-mainline/lk2nd/blob/6b85f2e755d09f95dd21950071850049e5b22b6c/dts/msm8916/msm8916-samsung-r04.dts#L44-L50

I think there should be a "DTB" line on the screen that would show which file to pick.

Ilya114, 2024-01-07:
> if your device shows "model = unknown" on the screen, you will need to add it to the lk2nd's device database.
>
> You would need to add something like this to a correct [...]

Compiled with

```
j5xn3g {
		model = "Samsung Galaxy J5 2016 (SM-J510H)";
		compatible = "samsung,j5xn3g", "qcom,msm8916", "lk2nd,device";
		lk2nd,match-bootloader = "J510H*";

		#include "msm8916-samsung-j5.dtsi"
	};
```

and have stucked on logo

TravMurav, 2024-01-07:
If you compile lk2nd without any changes, does it work?

Could you tell me what was in the DTB line on screen and what file did you add the above to?

Ilya114, 2024-01-07:
> does it work?

No

TravMurav, 2024-01-07:
How do you build and install the lk2nd image?

Ilya114, 2024-01-07:
> How do you build and install the lk2nd image?

`make TOOLCHAIN_PREFIX=arm-linux-gnueabihf- lk2nd-msm8916`

`heimdall flash --BOOT lk2nd.img`

Ilya114, 2024-01-07:
Maybe should I compile 64 bit?

TravMurav, 2024-01-07:
Can you get an `arm-none-eabi-` toolchain? lk2nd is always 32bit but I won't be surprised if it won't build properly with a hf toolchain...

Ilya114, 2024-01-07:
> Can you get an `arm-none-eabi-` toolchain? lk2nd is always 32bit but I won't be surprised if it won't build properly with a hf toolchain...

So it works, let me compile with 

```
j5xn3g {
		model = "Samsung Galaxy J5 2016 (SM-J510H)";
		compatible = "samsung,j5xn3g", "qcom,msm8916", "lk2nd,device";
		lk2nd,match-bootloader = "J510H*";

		#include "msm8916-samsung-j5.dtsi"
	};
```

Ilya114, 2024-01-07:
@TravMurav Model it still unknown

TravMurav, 2024-01-07:
> Could you tell me what was in the DTB line on screen and what file did you add the above to?

then let's continue with the second question :)

Ilya114, 2024-01-07:
> > Could you tell me what was in the DTB line on screen and what file did you add the above to?
> 
> then let's continue with the second question :)

`DTB - <248 0xf808ff01 0 0x5>`

I put in `dts/msm8916/msm8916-samsung-r04.dts`

TravMurav, 2024-01-07:
the dtb line says 248/.../5 so your file is `dts/msm8916/msm8216-samsung-r05.dts`. (note the 8**2**16 for 3g-only SoC)

You can find those numbers in that file, this is why I told you to check `dts/msm8916/msm8216-samsung-r<XX>.dts`, sorry that I wasn't clear enough

Ilya114, 2024-01-07:
> the dtb line says 248/.../5 so your file is `dts/msm8916/msm8216-samsung-r05.dts`. (note the 8**2**16 for 3g-only SoC)
> 
> You can find those numbers in that file, this is why I told you to check `dts/msm8916/msm8216-samsung-r<XX>.dts`, sorry that I wasn't clear enough

So now all works fine, I will make a PR, thanks!

PR #319 (2024-01-07): Add Ukrainian version of Samsung Galaxy J5 2016
Fixes #318
closed 2024-01-07 by TravMurav