msm8916-mainline/lk2nd #323: dts: add Samsung Galaxy Grand Max grandmaxltekx (SM-G720N0)

HPotty36, 2024-02-10:
I could flash the lk2nd image file on my SM-G720N0, but the fastboot commands don't work at all. Please add support for the Korean variants. Thanks in advance.

TravMurav, 2024-02-12:
Hi!

You would need to add your device model to lk2nd similar to this:

https://github.com/msm8916-mainline/lk2nd/blob/e7eb0707d6ef3723a310bda7962414dac7c9035a/dts/msm8916/msm8916-samsung-r04.dts#L79-L88

You should see a DTB line on screen, use the last value it shows to know which file to add it to. I.e. if it says `0x04` then you add to `msm8916-samsung-r04.dts`.

wonderfulShrineMaidenOfParadise, 2024-02-15:
Cc @third-tester

wonderfulShrineMaidenOfParadise, 2024-02-28:
There is already a dts done, but somehow not submitted.
You can also squash, clean up, rebase and then push the branch.

https://github.com/third-tester/lk2nd/tree/grandmaxltekx

wonderfulShrineMaidenOfParadise, 2024-06-12:
Consider joining [#msm8916-mainline:matrix.org](https://matrix.to/#/#msm8916:postmarketos.org) for faster helps.