msm8916-mainline/lk2nd #324: lk2nd not booting SM-A500FU

DaTrafficCone, 2024-02-15:
Hi, i'm trying to install postmarketOS on my Galaxy A5 (2015) and i'm trying to install lk2nd through heimdall and it will not boot to lk2nd. If i try holding vol up + power it just restarts.

TravMurav, 2024-02-19:
Hi! Could you explain how exactly did you flash lk2nd? i.e. which lk2nd build did you use and which commands did you issue to flash it?

wonderfulShrineMaidenOfParadise, 2024-02-21:
Not every SM-A500 is FU. Try getting into heimdall to check the model.

wonderfulShrineMaidenOfParadise, 2024-02-21:
Consider joining [#msm8916-mainline:matrix.org](https://matrix.to/#/#msm8916:postmarketos.org) for faster helps.

DaTrafficCone, 2024-02-29:
> Hi! Could you explain how exactly did you flash lk2nd? i.e. which lk2nd build did you use and which commands did you issue to flash it?

I used Heimdall to flash it and it was the latest build

TravMurav, 2024-02-29:
> I used Heimdall to flash it and it was the latest build

I will assume you used `lk2nd.img`. Can you confirm the exact model of your device (SM-A500??)
I believe it should be written on the back of the housing, at the bottom. Likely also on the box.

TravMurav, 2024-02-29:
> My model is the A500FU

Can you try making sure your device runs the latest firmware from samsung and then try installing lk2nd again?

DaTrafficCone, 2024-02-29:
> > My model is the A500FU
> 
> 
> 
> Can you try making sure your device runs the latest firmware from samsung and then try installing lk2nd again?
> 
> 

I tried that, it used to run LineageOS 18 but i reflashed the latest stock back to it and it still wouldn't show.