msm8916-mainline/lk2nd #327: Wileyfox Swift not booting with `fastboot boot lk2nd-msm8916.img`

lesleyrs, 2024-02-21:
https://github.com/msm8916-mainline/lk2nd/blob/e7eb0707d6ef3723a310bda7962414dac7c9035a/dts/msm8916/msm8916-qrd9-v1.dts#L11

I see that it's in the dts file but  `fastboot boot lk2nd-msm8916.img` still gives me the error: `FAILED: (remote: 'dtb not found')

I'm not sure how to adapt it to make it work for my device, the bootloader doesn't show much info?

tuandatdavid, 2024-02-24:
install dtb

TravMurav, 2024-02-25:
Hi! Can you check if your device has cyanogen bootloader (in fastboot mode it shows a menu with either CYANOGEN or WILEYFOX logo on the top of the screen) or generic one (will see big green START similar to lk2nd)?

If it's the generic one, you might want to install an earlier firmware build of cyanogenos and try again

lesleyrs, 2024-02-25:
There's no logo in fastboot but it has the green START, it says `bootloader version - crackling-N-ga787dc4`.

I didn't realize having the latest android 7 update would break compatibility even though it's many years old now, so I should try this "nightly" android 6 build here? https://cyanogenmodroms.com/crackling/

TravMurav, 2024-02-26:
There are few devices with "updated" (outdated??) bootloader like this and the easiest way to get lk2nd booting is to install the cyanogen one from the official/"stock" Cyanogen OS rom package (not cyanogenmod custom rom). We had a link in pmOS wiki but apparently it got stale by now...

Alternatively we could try to figure out how to make this funny bootloader happy with lk2nd but this would be more involved

lesleyrs, 2024-02-27:
@TravMurav I got it working by downloading this fastboot package and running `./flash-radio.sh` from it: https://cyanogenupdatetracker.com/downloads/ So the androidfilehost link [here](https://wiki.postmarketos.org/wiki/Wileyfox_Swift_(wileyfox-crackling)) should be replaced with that one here (I don't have an account)

After going through that install, android wouldn't boot past the cyanogenmod ready logo (which was looking quite different than the stock android 7 I had, it used to be a white background with the logo now it's black.

However `fastboot flash boot lk2nd.img` was able to work immediately after that! Wifi does work although it's not saved across boots along with some other settings, and starting phone makes double buzz sound instead of 1. Thx for the help.

closed 2024-02-27 by lesleyrs

TravMurav, 2024-02-27:
Thanks! I've updated the link in the wiki.

> though starting phone makes double buzz sound instead of 1.

This is correct: First buzz is from the cyanogen bootloader and second is from lk2nd. Note that when you want to get to lk2nd fastboot mode (which is the one you'd want to use most of the time), you need to start pressing the vol- immediately after the first buzz. Otherwise the original bootloader will take you to recovery instead.