msm8916-mainline/lk2nd #331: `Error code 3039 at boot_elf_loader.c Line 1413` On UART after flashing ABOOT partition

ElectroBoy404NotFound, 2024-03-03:
Hi there,

I flashed the prebuilt MSM8916 on to my Lenovo A6000's ABOOT partition using edl tools.
Now I get no display and on my UART I get `Error code 3039 at boot_elf_loader.c Line 1413`
I'm unable to force the device into edl mode and it gets detected as `QHSUSB__BULK` on my linux PC. I don't know how to proceed from here.

Please Help me.

[image]

TravMurav, 2024-03-03:
Hi!

If you flashed lk2nd.img to `aboot`, you installed it incorrectly and destroyed your original android bootloader. lk2nd.img is supposed to be installed in `boot` instead.

I suggest you to restore `aboot` from the backup you made and install lk2nd correctly.

ElectroBoy404NotFound, 2024-03-03:
> Hi!
> 
> If you flashed lk2nd.img to `aboot`, you installed it incorrectly and destroyed your original android bootloader. lk2nd.img is supposed to be installed in `boot` instead.
> 
> I suggest you to restore `aboot` from the backup you made and install lk2nd correctly.

Ah, but how do I do that? Now even the USB drives aren't showing up

TravMurav, 2024-03-03:
Switch the device into EDL the same way you did before and flash aboot back.

ElectroBoy404NotFound, 2024-03-03:
> Switch the device into EDL the same way you did before and flash aboot back.

I used the button combo, now the combo does nothing

ElectroBoy404NotFound, 2024-03-03:
Also looks like I destroyed the ABOOT partition even more as now this is what I get:
[image]

ElectroBoy404NotFound, 2024-03-03:
Ok, finally after a lot of google, I found the EMMC pins and shorted the CLK pin to force the mobile into edl. Thank you for your help!

closed 2024-03-03 by ElectroBoy404NotFound