msm8916-mainline/lk2nd #350: [rebase/next] fairphone2 (msm8974) : crash when no screen plugged in

brockauly, 2024-04-10:
This has already been investigated with the help of @TravMurav (see Matrix chat March 29).
I get boot log from UART. Since no register trace is printed at crash, we used prints to identify the faulty function call.

Faulty line is [https://lxr.trvn.ru/lk2nd/rebase;;next/source/lk2nd/display/cont-splash/pipe.c#L79](https://lxr.trvn.ru/lk2nd/rebase;;next/source/lk2nd/display/cont-splash/pipe.c#L79).

I added a `return false` before [https://lxr.trvn.ru/lk2nd/rebase;;next/source/lk2nd/display/cont-splash/target_display.c#L54](https://lxr.trvn.ru/lk2nd/rebase;;next/source/lk2nd/display/cont-splash/target_display.c#L54) to make lk2nd boot correctly and I can use fastboot.

@TravMurav asked me to get some register values and found them interesting and said a proper solution could be found later.
`fastboot oem debug readl 0xfd8c2328` gives `0x80000000`.
`fastboot oem debug readl 0xfd8c2304` gives `0xa0222000`.

Log with DEBUG=2 :
```
Format: Log Type - Time(microsec) - Message
Log type: B - since boot(excluding boot rom).  D - delta
B -    134322 - SBL1, Start
B -    139507 - scatterload_region && ram_init, Start
D -        30 - scatterload_region && ram_init, Delta
B -    156221 - pm_device_init, Start
D -     27297 - pm_device_init, Delta
B -    183610 - boot_flash_init, Start
D -     12108 - boot_flash_init, Delta
B -    196145 - boot_config_data_table_init, Start
D -    134932 - boot_config_data_table_init, Delta
B -    332633 - sbl1_ddr_set_params, Start
B -    335164 - Pre_DDR_clock_init, Start
D -       274 - Pre_DDR_clock_init, Delta
D -         0 - sbl1_ddr_set_params, Delta
B -    346571 - Detected Samsung SDRAM, Start
B -    350689 - Detected Samsung SDRAM, Start
B -    355996 - pm_driver_init, Start
D -     26596 - pm_driver_init, Delta
B -    383995 - clock_init, Start
D -       183 - clock_init, Delta
B -    387075 - Image Load, Start
B -    409737 - Tz Execution, Start
D -    184952 - Tz Execution, Delta
B -    602802 - Image Load, Start
B -    613324 - Signal PBL to Jump to RPM FW
B -    613599 - sbl1_wait_for_ddr_training, Start
D -     31964 - sbl1_wait_for_ddr_training, Delta
B -    649833 - Image Load, Start
B -    650870 - WDT Execution, Start
D -       213 - WDT Execution, Delta
B -    655262 - Image Load, Start
B -    676276 - sbl1_efs_handle_cookies, Start
D -       152 - sbl1_efs_handle_cookies, Delta
B -    678198 - RPM sync cookie updated
B -    681827 - SBL1, End
Android Bootloader - UART_DM Initialized!!!
[290] Not able to search the panel:
[430] Panel CMD: command mode dma test failed
[430] dsi panel init error
[500] Not able to search the panel:
[640] Panel CMD: command mode dma test failed
[650] dsi panel init error
[720] Not able to search the panel:
[890] Panel CMD: command mode dma test failed
[940] Panel CMD: command mode dma test failed
[1010] Panel CMD: command mode dma test failed
[1080] Panel CMD: command mode dma test failed
[1140] Panel CMD: command mode dma test failed
[1210] Panel CMD: command mode dma test failed
[1280] Panel CMD: command mode dma test failed
[1340] Panel CMD: command mode dma test failed
[1410] Panel CMD: command mode dma test failed
[1510] Panel CMD: command mode dma test failed
[1610] dsi panel init error
[1620] Not able to search the panel:
[1680] Logo config doesn't match with fb config. Fall back default logo
[1740] Panel CMD: command mode dma test failed
[1740] dsi panel init error
[1750] partition misc doesn't exist
[1750] error in emmc_recovery_init
[1760] No 'misc' partition found
[1760] Error reading MISC partition
[1760] failed to get ffbm cookieAndroid Bootloader - UART_DM Initialized!!!
[0] welcome to lk

[0] calling constructors
[0] initializing heap
[0] SCM call: 0x2000601 failed with :fffffffc
[0] Failed to initialize SCM
[0] initializing threads
[0] initializing dpc
[0] initializing timers
[0] creating bootstrap completion thread
[10] top of bootstrap2()
[10] initializing platform
[10] platform_init()
[10] initializing target
[10] target_init()
[10]  Initializing MMC host data structure and clock!
[30] Decoded CID fields:
[30] Manufacturer ID: 15
[30] OEM ID: 0x100
[40] Product Name: BWBC3R
[40] Product revision: 0.2
[40] Product serial number: 103F620
[40] Manufacturing date: 2 1998
[50] Decoded CSD fields:
[50] cmmc_structure: 3
[50] card_cmd_class: f5
[50] write_blk_len: 512
[60] read_blk_len: 512
[60] r2w_factor: 4
[60] sector_size: 0
[60] c_size_mult:7
[60] c_size: 4095
[70] nsac_clk_cycle: 100
[70] taac_ns: 15000000
[70] tran_speed: 0 kbps
[70] erase_blk_len: 0
[70] read_blk_misalign: 0
[80] write_blk_misalign: 0
[80] read_blk_partial: 0
[80] write_blk_partial: 0
[80] wp_grp_size: 16384
[90] Card Capacity: 31268536320 Bytes
[90] MMC card: BWBC3R (0.2, 02 1998), manufacturer: 15, OEM: 100, capacity: 31268536320 bytes
[100] SDHC Running in HS400 mode
[150] Done initialization of the card
[160] erase_grpsize: 31
[160] erase_grpmult: 31
[160] wp_grpsize: 15
[160] wp_grpen: 1
[160] perm_wp: 0
[170] temp_wp: 0
[170] calling apps_init()
[170] lk2nd_init()
[170] Booted @ 0x8000, r0=0x0, r1=0x0, r2=0x1e00000
[180] Found valid DTB with 1317 bytes total
[180] Command line from previous bootloader: lk2nd androidboot.emmc=true androidboot.serialno=103f620 an:
[190] Failed to find matching lk2nd device node: -1
[210] GPIO 5 status is 1
[210] Display Init: Start
```

wonderfulShrineMaidenOfParadise, 2025-10-22:
I had similar issue but it seems to be fixed already. Could you try latest release?

Biswa96, 2025-10-22:
It'd be helpful if the docs mention what the appropriate value should be set with LK2ND_DISPLAY.