msm8916-mainline/lk2nd #367: Keys are working backwards

applestr1x, 2024-06-20:
Hello. I boot the image from fastboot mode, there is no problem, but after booting, I cannot start the process. I'm stuck on the start screen. When I press the volume up button it changes options. When I press the power button, it changes the option, but when I press the volume down button, it does not respond at all. Normally it should start when I press the power button. Can you fix the problem? Phone with MSM8939 chipset. Its brand is Vestel Venus v3 5570.

TravMurav, 2024-06-21:
Hi! This device probably have non-standard keys pins and it was never added to lk2nd, so lk2nd doesn't know how to handle it.
You will need to add a node for your device with the keys correctly defined, similar to this:
https://github.com/msm8916-mainline/lk2nd/blob/f64f709c3263b8935fed430ae326b7de3373e67a/lk2nd/device/dts/msm8916/msm8916-lg.dts#L11-L29

Note that gpio number values are device-specific, you may need to check downstream devicetree to find the correct ones.

You can find lk2nd specific dts documentation here: https://github.com/msm8916-mainline/lk2nd/blob/main/Documentation/dt-bindings.md

wonderfulShrineMaidenOfParadise, 2024-06-21:
Consider joining [#msm8916-mainline:matrix.org](https://matrix.to/#/#msm8916:postmarketos.org) for helps with porting.