msm8916-mainline/lk2nd #370: dts: msm8974: hammerhead: cannot boot lk2nd-msm8974.img 17.0

wonderfulShrineMaidenOfParadise, 2024-06-22:
On hammerhead, it cannot boot lk2nd-msm8974.img 17.0 after `fastboot flash boot lk2nd-msm8974.img` in stock fastboot.
It loads wrong DTB after `fastboot boot lk2nd-msm8974.img`, which is supposed to be appended DTB `victara.dtb`, and the model becomes `Unknown (FIXME!)`.

barni2000, 2024-10-02:
Possible workaround LK2ND_DTBS="" LK2ND_ADTBS='msm8974-lge-hammerhead.dtb'

barni2000, 2024-10-05:
> Possible workaround LK2ND_DTBS="" LK2ND_ADTBS='msm8974-lge-hammerhead.dtb'

another possible workaround is `LK2ND_QCDTBS=""`

barni2000, 2024-10-05:
> On hammerhead, it cannot boot lk2nd-msm8974.img 17.0 after `fastboot flash boot lk2nd-msm8974.img` in stock fastboot.
> 
> To workaround this, we need to drop `msm8974pro-ab-pm8941-mtp.dtb` and `msm8974pro-ac-pm8941-mtp.dtb` from `rules.mk`.

This workaround is not working anymore.

wonderfulShrineMaidenOfParadise, 2024-10-26:
> This workaround is not working anymore.

It still works on 19.0, but I see your fix working as well. Closing.

closed 2024-10-26 by wonderfulShrineMaidenOfParadise