msm8916-mainline/lk2nd #385: Unknown (FIXME!) in Galaxy A5000

bumagh, 2024-07-03:
I flash lk2nd to Galaxy A5000 by twrp (install img to BOOT) ,than reboot it show "Unknown (Fixme)

bumagh, 2024-07-03:
log 
`[0] welcome to lk

[10] platform_init()
[10] target_init()
[20] MMC card: R311MB (0.0, 01 2006), manufacturer: 15, OEM: 100, capacity: 15758000128 bytes
[30] SDHC Running in HS200 mode
[30] Done initialization of the card
[40] lk2nd_init()
[40] Booted @ 0x80008000, r0=0x0, r1=0x0, r2=0x81e00000
[40] Found valid DTB with 3745 bytes total
[50] Command line from previous bootloader: lk2nd androidboot.sec_atd.tty=/dev/ttyHSL0 loglevel=4 sec_log=0x100000@0xafe00008 sec_dbg=0x80000@0xaff00008 samsung.board_rev=10 androidboot.warranty_bit=1 androidboot.bootloader=A5000ZCU1BQB2 lpm_levels.sleep_disabled=1 fg_reset=0 androidboot.boot_recovery=0 androidboot.debug_level=0x4f4c sec_debug.enable=0 sec_debug.enable_user=0 androidboot.cp_debug_level=0x55FF sec_debug.enable_cp_debug=0 vmalloc=340M cordon=03efcd5726a8d00a824c395195b66b26 connie=SM-A5000_OPEN_CHN_0179d6c821000e880753737e61d75570 lcd_id=0x812003 androidboot.security_mode=1526595584 androidboot.emmc=true androidboot.serialno=xxxxxx androidboot.baseband=msm lcd_id=0x812003 mdss_mdp.panel=1:dsi:0:ss_dsi_panel_EA8061V_AMS497EE01_HD:1:none
[110] Failed to find matching lk2nd device node: -1
[130] MDP continuous splash detected: pipe RGB_0, base: 0x83200000, stride: 2160, src: 720x1280 (0,0), img: 720x1280, out: 720x1280 (0,0), format: 0x2243f (bpp: 3)
[140] Display refresh: cmd mode: 0, auto refresh: 0 (sel: 0x1f20)
[150] Registering wrapper bio devices...
[150] Registering mmc_sdhci bio devices...
[170] Error: Command timeout error
[170] Failure getting OCR response from MMC Card
[170] MMC card failed to respond, try for SD card
[180] Error: Command timeout error
[180] The response for CMD8 does not match the supplied value
[180] Failed to initialize SD card
[190] Failed detecting MMC/SDC @ slot2
[190] SD card MMC is unavailable.
[190] block devices:
[200]  | dev        | label           | size       | Leaf |
[200]  | wrp0p26    | lk2nd           |    512 KiB | Yes  |
[210]  | wrp0p25    | userdata        |  12251 MiB | Yes  |
[210]  | wrp0p24    | cache           |    200 MiB | Yes  |
[220]  | wrp0p23    | system          |   2384 MiB | Yes  |
[220]  | wrp0p22    | persdata        |      9 MiB | Yes  |
[230]  | wrp0p21    | persist         |      8 MiB | Yes  |
[230]  | wrp0p20    | ssd             |      8 KiB | Yes  |
[240]  | wrp0p19    | fsc             |      3 MiB | Yes  |
[240]  | wrp0p18    | backup          |      6 MiB | Yes  |
[250]  | wrp0p17    | fota            |     12 MiB | Yes  |
[250]  | wrp0p16    | recovery        |     15 MiB | Yes  |
[260]  | wrp0p15    | boot            |     12 MiB | Yes  |
[270]  | wrp0p14    | modemst2        |      3 MiB | Yes  |
[270]  | wrp0p13    | modemst1        |      3 MiB | Yes  |
[280]  | wrp0p12    | efs             |     14 MiB | Yes  |
[280]  | wrp0p11    | param           |     10 MiB | Yes  |
[290]  | wrp0p10    | pad             |     10 MiB | Yes  |
[290]  | wrp0p9     | sec             |     16 KiB | Yes  |
[300]  | wrp0p8     | fsg             |      3 MiB | Yes  |
[300]  | wrp0p7     | qhee            |    512 KiB | Yes  |
[310]  | wrp0p6     | qsee            |    512 KiB | Yes  |
[310]  | wrp0p5     | rpm             |    512 KiB | Yes  |
[320]  | wrp0p4     | aboot           |      2 MiB | Yes  |
[320]  | wrp0p3     | ddr             |     32 KiB | Yes  |
[330]  | wrp0p2     | sbl1            |    512 KiB | Yes  |
[330]  | wrp0p1     | modem           |     57 MiB | Yes  |
[340]  | wrp0p0     | apnhlos         |     15 MiB | Yes  |
[340]  | wrp0       |                 |  15028 MiB |      |
[350] boot: Trying to boot from the file system...
[360] ERROR: Invalid boot image header
[360] ERROR: Could not do normal boot. Reverting to fastboot mode.
[370] fastboot_init()
[470] USB init ept @ 0x8f68b000
[490] udc_start()
[590] -- reset --
[590] -- portchange --
[700] fastboot: processing commands
[127340] fastboot: oem log
`

closed 2024-07-04 by bumagh

wonderfulShrineMaidenOfParadise, 2024-07-04:
The issue is not really solved if it's still `Unknown (FIXME!)`.