msm8916-mainline/lk2nd #398: Support for msm8937

Minionguyjpro, 2024-08-06:
Hello, is there any chance to support this SoC family? I have an SDM 429 which is based off this chipset. I've looked a bit through the porting and it shouldn't be too hard (it seems). Can I however do msm8937 at all? It is listed in the porting as SoC based off the one wiki, so...

TravMurav, 2024-08-06:
Hi! Consider using msm8952 target: https://github.com/msm8916-mainline/lk2nd/tree/main/lk2nd/device/dts/msm8952

Minionguyjpro, 2024-08-06:
> Hi! Consider using msm8952 target: https://github.com/msm8916-mainline/lk2nd/tree/main/lk2nd/device/dts/msm8952

I'll try, thank you! Could you guide me through the process of porting if I need so? I will tell you whether I am getting somewhere!

TravMurav, 2024-08-06:
You're always free to ask questions in the chat

https://github.com/msm8916-mainline/lk2nd/blob/main/README.md?plain=1#L92