msm8916-mainline/lk2nd #426: Menu on startup?

TailsFanLOL, 2024-09-12:
Yo! Can you please add an option to show a menu on startup with multiple startup choices (main, the entry from the cfg file, recovery, fastboot, edl...)? The file/byte/bit where one could enable this could be appended to the lk2nd image or at least defined compile time. I am sorry if this is out of scope for this project and please suggest the closest one if it is (I use markw btw).

TailsFanLOL, 2024-09-12:
missclicked

TailsFanLOL, 2024-09-14:
Also, this should allow one to have multiple entries in the cfg file as well, and perhaps multiple kernel partitions

TravMurav, 2024-09-14:
Hi! I've resurrected a compile time flag to force entering the boot menu, it was missed in the rework sorry. See https://github.com/msm8916-mainline/lk2nd/commit/b96e23c0c41d33d446d8af4bf6963dc0844bbada

However providing interactive bootloader experience is out of scope for lk2nd. For that you're better getting a (close-to-) EBBR environment by i.e. porting u-boot to your platform. Chainloading u-boot from lk2nd to retain it's dt fixups is relatively simple assuming u-boot itself works on the given platform.

closed 2024-09-14 by TravMurav

TailsFanLOL, 2024-09-14:
Well, I don't see how this would be difficult to implement in any way. One could argue it's already an interactive bootloader experience ¯⁠\\⁠_⁠(⁠ツ⁠)⁠_⁠/⁠¯