msm8916-mainline/lk2nd #431: lk2nd.pass-ramoops is not working correctly on msm8956

barni2000, 2024-09-22:
In downstream dmesg ramoops is on different address what lk2nd detects and ```oem ramoops console``` is not working and ```oem ramoops raw``` gives back garbage.

```
(bootloader) CONSOLE: (0xfdb6febf) 0x3eff3fbb @ (0xbffc0000 + 0xbdbbff7d)
```

```
<6>[    0.000000] ramoops: msm_reserve_ramoops_memory addr=dfc00000,size=400000
<6>[    0.000000] ramoops: msm_reserve_ramoops_memory record_size=0,ftrace_size=0
```

TravMurav, 2024-09-22:
I'm afraid at the moment the ramoops tool in lk2nd is completely ineffective for downstream linux. There is no way to actually "detect" ramoops settings so the lk2nd approach is to:

- precompute a reasonably reliable memory region and settings based on the platform (end of scratch region);
- upon boot (if booting mainline), update the devicetree and add those precomputed settings;
- on next boot, compute the settings once again, getting the same address and use that to read ramoops.

Obviously this doesn't work for downstream and I i.e. don't know if it's safe to change ramoops node settings for old kernels even. For that reason I've decided that the scope of the ramoops tool should be limited to mainline linux only, as for downstream one usually can use recovery to dump ramoops, likely also benefiting from ECC and compression which lk2nd at the moment cant decode.

barni2000, 2024-09-22:
> I'm afraid at the moment the ramoops tool in lk2nd is completely ineffective for downstream linux. There is no way to actually "detect" ramoops settings so the lk2nd approach is to:
>
>     * [...]

I would use it for mainline. I have just mentioned downstream because of the used region.

TravMurav, 2024-09-22:
so it's broken when booting with mainline? Does the region gets (somewhat) cleared on reboot if you boot with lk2nd.pass-ramoops=zap?

barni2000, 2024-09-22:
> so it's broken when booting with mainline? Does the region gets (somewhat) cleared on reboot if you boot with lk2nd.pass-ramoops=zap?

Yes, i have got this with zap:
```
❯ fastboot oem ramoops regions
(bootloader) ramoops regions:
(bootloader) DUMP:
(bootloader) CONSOLE: (0x0) 0x0 @ (0xbffc0000 + 0x0)
```

TravMurav, 2024-09-22:
then I'd imagine lk2nd side at least passes the ramoops node to mainline linux (and clears the memory for it with =zap) so now I'd assume the question is whether linux reads ramoops node or not...

barni2000, 2024-09-22:
```[    0.000000] OF: reserved mem: 0x00000000bff80000..0x00000000bfffffff (512 KiB) map non-reusable ramoops@dfc00000```

TravMurav, 2024-09-22:
But does the ramoops driver load and register the console device though?

barni2000, 2024-09-22:
```
		ramoops@dfc00000 {
			ecc-size = <0x00>;
			ftrace-size = <0x00>;
			compatible = "ramoops";
			reg = <0x00 0xbff80000 0x00 0x80000>;
			record-size = <0x2000>;
			console-size = <0x40000>;
			pmsg-size = <0x00>;
		};
		```

TravMurav, 2024-09-22:
maybe because your original dts already has that node? lk2nd tries to update existing ramoops node instead of creating a new one so they don't conflict

barni2000, 2024-09-22:
> maybe because your original dts already has that node? lk2nd tries to update existing ramoops node instead of creating a new one so they don't conflict
> 
> (...) in this case the name will be kept since the node would need to be deleted and recreated otherwise

my bad, https://github.com/msm8956-mainline/linux/commit/00d61070699c83a72d4f2abd893831a821fab30f#diff-6acc273965d0cc4d7c807c1959eb64ab92c40c7aa31a845aed986aa5c8e98917R89

barni2000, 2024-09-22:
It seems the issue will be the kernel has not got enough time to save logs during a crash.

barni2000, 2024-09-22:
the node seems fine now
```
		ramoops@bff80000 {
			ecc-size = <0x00>;
			pmsg-size = <0x00>;
			ftrace-size = <0x00>;
			record-size = <0x2000>;
			console-size = <0x40000>;
			reg = <0x00 0xbff80000 0x00 0x80000>;
			compatible = "ramoops";
		};

TravMurav, 2024-09-22:
I assume there is no issue left on lk2nd side then? Feel free to re-open and lmk if there is anything left

closed 2024-09-22 by TravMurav