msm8916-mainline/lk2nd #446: Huawei bootloader stange problem

hicode002, 2024-11-17:
I have a Huawei T1-A21L equipped with MSM8916, and it can be confirmed that secure boot is not enabled. However, Huawei's native firmware is 32-bit. I compiled LK1st and replaced the native firmware with 64 bit TZ and qhypstub. However, Huawei's SBL has additional signature verification for Aboot (not a Qualcomm certificate), so I had to use DB410C's SBL to boot into LK1st. Despite porting the panel, the screen did not respond and could not boot into the kernel, and the LED light did not light up (even though I had edited gpio). I suspect it may be related to the replacement of sbl, so I conducted another experiment using db410c sbl, and then using Huawei native tz hyp aboot. At this time, the device can enter fastboot normally, but the screen and LED lights are not on, and it cannot enter the kernel.

So what role did SBL play? Does sbl have an effect on screen or gpio initialization? What should I do? The compilation command for lk1st that I am using is
```
make TOOLCHAIN_PREFIX=arm-none-eabi- lk1st-msm8916 \
     LK2ND_BUNDLE_DTB=msm8916-huawei-hwt1aw2l.dtb \
     LK2ND_DISPLAY=boe_nt51017_10_800p_video
```
I have added the panel files into lk2nd/display/generated
help me!

stephan-gh, 2024-11-17:
> So what role did SBL play? Does sbl have an effect on screen or gpio initialization?

SBL turns on the regulators (= power supplies). If you need any GPIOs or similar, or additional non-standard power supplies then lk1st needs to implement that.

For example, huawei-g7 has a tps65132 that is needed for the panel to work and I had to add extra code for that: https://github.com/msm8916-mainline/lk2nd/pull/164/commits/bf8bccd8d67617209afab94ac60e35005d9f11df  
Another thing that is often missing is the brightness setting: https://github.com/msm8916-mainline/lk2nd/pull/164/commits/930590a74814ed19d76ac10bfeac6ab6513e7429

You need to provide more information about your setup (e.g. upload your changes you made for your devices).

Also, what does "cannot enter the kernel" mean? Is there a specific error?

hicode002, 2024-11-17:
> > 那么 SBL 扮演了什么角色呢？sbl 对屏幕或 gpio 初始化有影响吗？
> 
> SBL 打开稳压器（= 电源）。如果您需要任何 GPIO 或类似产品，或其他非标准电源，那么 lk1st 需要实现。
> 
> 例如，huawei-g7 有一个面板工作所需的 tps65132，我不得不为此添加额外的代码：[bf8bccd](https://github.com/msm8916-mainline/lk2nd/commit/bf8bccd8d67617209afab94ac60e35005d9f11df) 另一件经常丢失的事情是亮度设置：[930590a](https://github.com/msm8916-mainline/lk2nd/commit/930590a74814ed19d76ac10bfeac6ab6513e7429)
> 
> 您需要提供有关设置的更多信息（例如，上传您对设备所做的更改）。
> 
> 另外，“cannot enter the kernel” 是什么意思？是否有特定错误？

it just got stuck in bootloop.reboot again and again

hicode002, 2024-11-17:
> > So what role did SBL play? Does sbl have an effect on screen or gpio initialization?
> 
> SBL turns on the regulators (= power supplies). If you need any GPIOs or similar, or additional non-standard power supplies then lk1st needs to implement that.
> 
> For example, huawei-g7 has a tps65132 that is needed for the panel to work and I had to add extra code for that: [bf8bccd](https://github.com/msm8916-mainline/lk2nd/commit/bf8bccd8d67617209afab94ac60e35005d9f11df) Another thing that is often missing is the brightness setting: [930590a](https://github.com/msm8916-mainline/lk2nd/commit/930590a74814ed19d76ac10bfeac6ab6513e7429)
> 
> You need to provide more information about your setup (e.g. upload your changes you made for your devices).
> 
> Also, what does "cannot enter the kernel" mean? Is there a specific error?

不，我的设备明确不使用tps65132，而是在设备树里直接disable了tps65132.相反，我在设备树里看到了huawei自己的实现，huawei,xxx.

你可以在这里看我的设备的原生32位内核源代码，https://github.com/hicode002/android_kernel_huawei_hwt1a21w/blob/main/arch/arm64/boot/dts/qcom/huawei_msm8916_t110_p0_a21w/huawei_t110_p0_a21w.dts

hicode002, 2024-11-18:
> SBL turns on the regulators (= power supplies). If you need any GPIOs or similar, or additional non-standard power supplies then lk1st needs to implement that.
>
> For example, huawei-g7 has a tps65132 [...]

i want to ask you a question,can huawei g7 boot with 64-bit tz/hyp and origin huawei aboot(not lk1st)? I also tried to use Huawei's native Aboot with 64 bit TZ and HYP, but it cannot boot, the screen is black, there is no vibration, and the LED is not working.