msm8916-mainline/lk2nd #474: 4G LTE Router (MF968) Stuck in Boot Loop After Bootloader Change

Amm1rr, 2025-01-03:
### Description
After replacing the original bootloader with `lk2nd` on my Qualcomm 4G LTE Mobile WiFi Router (MF968), the device is now stuck in a boot loop, showing only the initial logo indefinitely. It is not recognized by my computer (no detection in `lsusb`), and attempts to enter EDL mode have failed.

### Device Details
- **Original OS:** Android  
- **Current Bootloader:** I'm not sure - lk2nd  
- **Model:** MF968  
- **Chipset:** msm8916  
- **Vendor/Product IDs:** 05c6:9024  
- **LTE Bands:** B1/B3/B5/B7/B8/B20/B38/B40/B41  
- **Hardware Version:** HW1.2  
- **Software Version:** HMIFI_V29_CMX3_S1_EU_240806  

### Current State
- Boot loop at the initial logo.  
- Not detected by computer when connected via USB.  
- No entry in `lsusb` or response from the device when using a USB cable.  
- Requires battery removal to power off.  

### Steps Tried
1. Replaced the bootloader with `lk2nd`.  
2. Attempted to connect the device to a computer (no detection).  
3. Tried to enter EDL mode:
   - Shorted black and green cables for 5 seconds (unsuccessful).  

### Questions
1. How can I recover the device from the boot loop?  
2. Is there a reliable way to verify correct EDL mode entry?  
3. Are there alternative recovery methods or tools specific to Qualcomm devices for this issue?  

Any guidance or suggestions to restore the device or revert the bootloader would be appreciated.

**Thank you!**

TravMurav, 2025-01-04:
Hi! Could you clarify the exact steps you performed to "replace the original bootloader with lk2nd"? Have you preformed something similar to the guide at https://wiki.postmarketos.org/wiki/Zhihe_series_LTE_dongles_(generic-zhihe) or some different steps? Listing all the steps and commands in order would be ideal.

if you've corrupted your firmware, the only recovery you will have is entering EDL (via shorting usb lanes or some testpoint specific to the board) and flashing a backup of the firmware you've surely made. (or fixing the mistake made when originally installing alternative firmware to make the device bootable)

Amm1rr, 2025-01-04:
Thank you for your response.

Honestly, it was my own negligence. Initially, I rooted the device using **ADB** and the **SuperSU** method. However, due to the modem's very small display, I couldn’t press the “Allow” option on the screen. I decided to **unroot** it, but the script I used caused the device to enter a **restart loop**. At this stage, the modem was still detectable by the computer.

To resolve the issue, I attempted to install **lk2nd** via **fastboot** using the command:

```
fastboot flash boot lk2nd.img
```

After flashing, the modem started showing its logo indefinitely and is no longer recognized by the computer, not even in fastboot mode.

I still haven’t figured out what model my board is (**HM-MIFIB-V12**), which is why using EDL hasn’t worked for me so far. I realize I have an issue with my approach in EDL mode and need to investigate further. I thought perhaps someone might be able to provide a clue or some guidance.

[image]

TravMurav, 2025-01-05:
> `fastboot flash boot lk2nd.img`
> After flashing, the modem started showing its logo indefinitely and is no longer recognized by the computer, not even in fastboot mode.

My guess would be that one of:
1) This device is not actually based on msm8916 (please confirm this by removing the metal shields and checking the SoC is labelled "MSM8916", additionally providing photos of both sides, like above, but with shields removed would be ideal. I hope those just snapped on)
2) This device is msm8916 but the bootloader expects some different board id for the boot image, which was never added to lk2nd ("MF968" is not in the supported list, though some other similar devices are)

Considering you've only replaced the `boot` partition, you should still be able to boot into fastboot mode as your original bootloader should be intact (in fact for qcom bootloader this should be default for unbootable kernel image). If you haven't tried to do so, I'd suggest trying to press that button on the photo before plugging the device in and seeing if it goes to fastboot/edl. There are also some uart testpoints on the top middle, those could help in understanding what the device does.

185264646, 2025-02-15:
It doesn't seem MF968 is using MSM8916. See https://detail.1688.com/offer/787999782315.html#offer-title-1081181309894-1, It's using a 9607 according to the site.

afaik, only mf32xx and mf601xx series boards are based on msm8916. mf7xx is using some ZXIC or ASR SoCs with completely different softwares. Please try `adb reboot edl` and see if you can get into EDL mode

Cc: @TravMurav

Amm1rr, 2025-02-16:
Thank you. I also think you are right.
Unfortunately, I haven't been able to remove the metal shield yet. As soon as I manage to open it, I'll provide more information.
Currently, **the Loading** logo is displayed on the LCD, and it is not recognized by **lsusb**. It doesn't respond to any commands.