msm8916-mainline/lk2nd #497: msm8960 / lk2nd/msmhtc,m7 (APQ8064) / doesnt works

lm2lm2, 2025-02-11:
hello

after several attempts, the lk2nd can't run on htc-m7

msm8960 / lk2nd/msmhtc,m7 (APQ8064) 

i dont know why, i can provide logs or attempt others manipulations

thanks !

user0-07161, 2025-02-11:
please be more specific; does it boot? if yes, please send logs. `fastboot oem log`; thanks.

lm2lm2, 2025-02-11:
> please be more specific; does it boot? if yes, please send logs. `fastboot oem log`; thanks.

went never up to get anything from fastboot commands.
only fastboot flash plus fastboot boot, in that case : twrp flash well at recovery, none went up to be fastboot boot, and none went up to be started even from taking part of recovery partition.

i dont know what's happening, even with the python script to create an image from both twrp recovery + lk2nd (8960 img) merger to final.img didnt worked.

user0-07161, 2025-02-12:
i'm honestly not sure if apq8064 works out of the box