msm8916-mainline/lk2nd #498: UFI devices: GPIO config seems wrong

185264646, 2025-02-16:
For UFI001c and jz0145. The device tree configured the EDL button to be pull up and active_low, which is likely to be wrong, and is different with zhihe-various node.

https://github.com/msm8916-mainline/lk2nd/blob/6a205e5317e11471e1b9eb2d807aa48d8155271f/lk2nd/device/dts/msm8916/msm8916-512mb-mtp.dts#L70

should be ACTIVE_HIGH and pull down as already defined in zhihe-various node.

@TravMurav

@rcky844 Have you tested the key before?

rcky844, 2025-02-17:
This is interesting, because EDL button was working for me.

I can't confirm it now that my unit is broken, so if most people's JZ0145 doesn't work, you can fix it. But please do not blindly fix it by following zhihe-various.

185264646, 2025-02-18:
> This is interesting, because EDL button was working for me.
> 
> I can't confirm it now that my unit is broken, so if most people's JZ0145 doesn't work, you can fix it. But please do not blindly fix it by following zhihe-various.

It's inconsistant with https://lore.kernel.org/all/20250114-xiaoxun-jz0145-v33-v2-3-495e986579ce@gmail.com/ . Please check it. Either lk2nd or linux dts must be wrong. The button can not be active high in linux while active low in lk2nd. How did you verify that the button works in lk2nd?