msm8916-mainline/lk2nd #519: Without the display panel connected, Xiaomi-land (MSM8937) Device goes to Qdloader mode on flashing lk2nd-msm8952.img. I NEED HEADLESS LK2ND build.

aakash-pro, 2025-04-17:
HI GUYS , I have Redmi 3s ( codename: xiaomi-land) chipset: MSM8937, bootloader is unlocked &
The display is not connected to the motherboard.

Booting the device with fastboot flash boot lk2nd-msm8952.img crashes the device into EDL mode. 

**I am using msm8952 image file**.

However the stock boot.img works perfect without display connected, fastboot commands works also fine on stock bootloader image. TWRP also boots fine with all adb commands working. The issue is only with lk2nd.

I think the LK2ND is crashing during display panel initialization . Can anyone please suggest me how to build lk2nd without display panel initialization code or bypass it completely. 

I want a headless Lk2nd !!

aakash-pro, 2025-04-18:
Exploring the code base , if found a flag BOOTLOADER_DISABLE_DISPLAY setting it to "true" shall bypass the display initialization code.

Full command

make TOOLCHAIN_PREFIX=arm-none-eabi- lk2nd-msm8952 LK2ND_DTBS="" LK2ND_ADTBS="msm8953-xiaomi-land.dtb" BOOTLOADER_DISABLE_DISPLAY=true

aakash-pro, 2025-04-22:
**UPDATE** : At last I have purchased & the connected the display to motherboard. Now LK2ND is working fine.

closed 2025-04-30 by aakash-pro

wonderfulShrineMaidenOfParadise, 2025-10-22:
If you are interested, `LK2ND_DISPLAY` is implemented in #363, which by default is set to `cont-splash`.
You can disable/unset it by passing `LK2ND_DISPLAY=` when building, and `cont-splash` will be disabled.