msm8916-mainline/lk2nd #523: Add support for Redmi 8A(olive)

HuangHuangHuHu66, 2025-04-25:
Please forgive me for just getting started with this. I have already read the **dt-bindings.md**, but I still have some questions that I don’t understand.
I'm running Redmi 7(pine) on Redmi 8A(olive), it doesn't work, even though they have the same SOC, I tried to compile lk2nd for Redmi 8A(olive), but I don't know how to do it. I don't know how to get msm-id, board-id, display related parameters.
Redmi 7(pine): https://github.com/msm8916-mainline/lk2nd/blob/main/lk2nd/device/dts/msm8952/sdm439-xiaomi-pine.dts
Redmi 8A(olive): https://github.com/MiCode/Xiaomi_Kernel_OpenSource/blob/olive-q-oss/arch/arm64/boot/dts/qcom/sdm439-olive.dts
https://github.com/MiCode/Xiaomi_Kernel_OpenSource/blob/olive-q-oss/arch/arm64/boot/dts/qcom/sdm439-olive.dtsi
https://github.com/redmi8a/android_device_xiaomi_olivelite

barni2000, 2025-04-28:
You will need custom minimal dtbo image for 8A.

HuangHuangHuHu66, 2025-04-29:
> You will need custom minimal dtbo image for 8A. board-id can be find in the original dtbo.img, you can use binwalk for extract dtbs from it

Thank you so much for your guidance.I used binwalk extract on the official dtbo.img and got a lot of dts. I tried to use different board-ids from them to create a lk2nd.img, but the lk2nd didn't work. I think I don't understand the custom minimal dtbo image for 8A. Can you explain the approach in more detail?

HuangHuangHuHu66, 2025-04-29:
> You will need custom minimal dtbo image for 8A. board-id can be find in the original dtbo.img, you can use binwalk for extract dtbs from it

I checked the board-id of LineageOS's dtbo.img and found two entries:
model = "PINE QRD";
qcom,board-id = <0x0b 0x02>;
model = "Olive QRD";
qcom,board-id = <0x0b 0x08>;
For the lk2nd project's pine: [lk2nd/lk2nd/device/dts/msm8952/sdm439-xiaomi-pine.dts at main · msm8916-mainline/lk2nd](https://github.com/msm8916-mainline/lk2nd/blob/main/lk2nd/device/dts/msm8952/sdm439-xiaomi-pine.dts)
I'm unsure how to match these board-ids.

HuangHuangHuHu66, 2025-05-01:
> You will need custom minimal dtbo image for 8A. board-id can be find in the original dtbo.img, you can use binwalk for extract dtbs from it

I worked on the dtbo project for Redmi 7A (pine) from this repository:https://github.com/barni2000/dtbo-lk2nd. I successfully modified and compiled the dtbo.img for Redmi 8A (olive), and now the lk2nd for Redmi 8A (olive) is running properly. Thanks again for your help.

barni2000, 2025-05-01:
> I worked on the dtbo project for Redmi 7A (pine) from this repository:https://github.com/barni2000/dtbo-lk2nd. I successfully modified and compiled the dtbo.img for Redmi 8A (olive), and now the [...]

Feel free open a PR (barni2000/lk2nd-dtbo) for collect the dtbo-s at same place

HuangHuangHuHu66, 2025-05-01:
> > > You will need custom minimal dtbo image for 8A. board-id can be find in the original dtbo.img, you can use binwalk for extract dtbs from it
> > 
> > 
> > I worked on the dtbo project for Redmi 7A (pine) from this repository:https://github.com/barni2000/dtbo-lk2nd. I successfully modified and compiled the dtbo.img for Redmi 8A (olive), and now the lk2nd for Redmi 8A (olive) is running properly. Thanks again for your help.
> 
> Feel free open a PR (barni2000/dtbo-lk2nd) for collect the dtbo-s at same place

model = "PINE QRD";
qcom,board-id = <0x0b 0x02>;
model = "Olive QRD";
qcom,board-id = <0x0b 0x08>;

HuangHuangHuHu66, 2025-05-01:
> Feel free open a PR (barni2000/dtbo-lk2nd) for collect the dtbo-s at same place

Sorry, I don't understand what "collect the dtbo-s" means. I checked the board-id of LineageOS's dtbo.img and found two entries:
model = "PINE QRD";
qcom,board-id = <0x0b 0x02>;
model = "Olive QRD";
qcom,board-id = <0x0b 0x08>;
I obtained the board-id from LineageOS's dtbo.img and it is correct.

barni2000, 2025-05-01:
> Sorry, I don't understand what "collect the dtbo-s" means. I checked the board-id of LineageOS's dtbo.img and found two entries: model = "PINE QRD"; qcom,board-id = <0x0b 0x02>; model = "Olive QRD"; [...]

It means the configs should be live in 1 repo instead of multiple separated repo.