msm8916-mainline/lk2nd #535: dts: msm8916-ee-harrier-mini: Warning

wonderfulShrineMaidenOfParadise, 2025-05-17:
```
compiling lk2nd/device/dts/msm8916/msm8916-ee-harrier-mini.dts
lk2nd/device/dts/msm8916/msm8916-ee-harrier-mini.dts:32.13-33.4: Warning (unit_address_vs_reg): /soc@0: node has a unit name, but no reg or ranges property
lk2nd/device/dts/skeleton64.dtsi:11.11-14.4: Warning (unique_unit_address): /memory@0: duplicate unit-address (also used in node /soc@0)
```

Since a96d4f3152ab41df51de385917a55abab1a03178 #531

TravMurav, 2025-05-17:
Seems like dtc can't be configured to treat warnings as errors... Annoying...

Pushed a few changes to suppress those warnings: 

- https://github.com/msm8916-mainline/lk2nd/commit/d087ca5eb83cc79ef7d2d4dea7a8ef2f305804c9
- https://github.com/msm8916-mainline/lk2nd/commit/8a31de8c81d8a13a7de4f76c1f7a2fc315d567be

+CC @halal-beef

halal-beef, 2025-05-17:
Ah, very sorry about that, must've missed those when building

halal-beef, 2025-05-17:
Phone still boots lk2nd with the changes

wonderfulShrineMaidenOfParadise, 2025-07-06:
Closing as the warnings are gone.

closed 2025-07-06 by wonderfulShrineMaidenOfParadise