msm8916-mainline/lk2nd #551: Motorola Moto G7 Power (ocean) bootloops after flashing lk2nd

unilock, 2025-06-13:
I'm attempting to install postmarketOS on my Motorola Moto G7 Power (ocean), which in turn requires flashing lk2nd. However, after flashing the latest release for msm8953 (v20.0 at the time of writing) and rebooting, the device bootloops - more specifically, it reaches the "Bootloader is unlocked" warning screen, waits five seconds, flashes black for a fraction of a second, then returns to the warning screen, repeating this process until the device is force-rebooted into the stock fastboot mode.

Holding volume up / down after the unlocked bootloader warning screen appears (to attempt to enter lk2nd's fastboot / recovery) makes no difference.

***

The output of `fastboot flash boot_a lk2nd-msm8953.img`:

```
 (bootloader) is-logical:boot_a: not found
 (bootloader) is-logical:boot_a: not found
 Sending 'boot_a' (258 KB)                          OKAY [  0.006s]
 Writing 'boot_a'                                   (bootloader) Image not signed or corrupt
 OKAY [  0.058s]
```

(supposedly, the "Image not signed or corrupt" message is normal)

I'm not sure how to get any further diagnostic information.

***

Also paging @ultra-azu, as she's the postmarketOS maintainer for this device.

XerTheSquirrel, 2025-06-26:
This occurs for me as well, after writing the latest it will boot loop showing the charging system as it is plugged in. I can get back to the bootloader flashing by holding volume down + power.

stephan-gh, 2025-06-30:
Is there an older version of lk2nd that is working correctly (preferably one of the releases from this repository, not just the really old version from msm8953-mainline/lk2nd)

XerTheSquirrel, 2025-07-19:
> > ... I had luck with `msm8953-mainline/lk2nd` which booted successfully, ...
> 
> Is there any way you could tell me how you managed that? Tried to compile it myself (following the pmOS wikipage) but it gives me numerous errors...

The errors are due to GCC being more strict these days, therefor I fixed the errors which would have previously only been warnings.

tulpenkiste, 2025-07-20:
> The errors are due to GCC being more strict these days, therefor I fixed the errors which would have previously only been warnings.

I don't know how to fix them myself 3:

barni2000, 2025-08-10:
ocean is sdm632 device and it will need custom empty/minimal dtbo image like xiaomi-onclite

tulpenkiste, 2025-08-14:
@barni2000 I probably got the dtbo files wrong. Also attached the dts that my phone uses from lineageOS as a reference

[original.dts.txt](https://github.com/user-attachments/files/21774588/original.dts.txt)

[sdm632-motorola-ocean.dts.txt](https://github.com/user-attachments/files/21774601/sdm632-motorola-ocean.dts.txt)

[dtboimg-motorola-ocean.cfg.txt](https://github.com/user-attachments/files/21774603/dtboimg-motorola-ocean.cfg.txt)

barni2000, 2025-08-14:
maybe device dt should move under QCDTBS list in makefile like at potter.

barni2000, 2025-08-16:
I have checked `twrp.img` it seems the device is using appended dtbs, maybe it is interfering with other devices so try `make TOOLCHAIN_PREFIX=arm-none-eabi- LK2ND_QCDTBS="" LK2ND_DTBS="" LK2ND_ADTB="sdm632-motorola-ocean.dtb" lk2nd-msm8953` also you will need that `dtbo.img` but you can try erase `dtbo` partition first.

tulpenkiste, 2025-08-16:
> I have checked `twrp.img` it seems the device is using appended dtbs, maybe it is interfering with other devices so try `make TOOLCHAIN_PREFIX=arm-none-eabi- LK2ND_QCDTBS="" LK2ND_DTBS="" [...]

Doing this results in a .img that cannot be flashed.
```
(bootloader) is-logical:boot_b: not found
(bootloader) is-logical:boot_b: not found
Sending 'boot_b' (300 KB)                          OKAY [  0.007s]
Writing 'boot_b'                                   FAILED (Status read failed (No such device))
fastboot: error: Command failed
```

valpackett, 2025-11-23:
Hi, I've just been bringing up river (ocean's less power-ful cousin, hah) and encountered the same problem.

Looks like the [dtbo source](https://github.com/barni2000/dtbo-lk2nd/blob/21fec24f22944b8dc490148266fb76177b80ef73/sdm632-motorola-ocean.dts) with multiple board IDs is completely bogus. The loader only looks at the **first** two numbers. I discovered this by making the same type of big list for river ­— it didn't work.. so I started commenting out the IDs.. and after commenting out two I booted into lk2nd!

```dts
/dts-v1/;

/ {
	qcom,board-id = //<0x145 0xa100>,
			// <0x145 0xa200>,
			<0x45 0xa100>,
			<0x45 0xa200>,
			<0x45 0xb1b0>;
	model = "river";

	__fixups__ {};
};
```

So that's been working on *one* ocean variant, and not any of the others.

The original dtbo partition contains multiple independent *whole* dtbs, each with just one board ID, not one dtb with a long long board-id.

tulpenkiste, 2025-11-23:
> ... please try [barni2000/dtbo-lk2nd#6](https://github.com/barni2000/dtbo-lk2nd/pull/6) ­— or directly my DTBO build: [dtbo-motorola.zip](https://github.com/user-attachments/files/23693725/dtbo-motorola.zip)

Will be trying! Going to refresh postmarketOS first...

barni2000, 2026-01-02:
New dtbo image is released this ticket can be closed.