msm8916-mainline/lk2nd #559: Moto G5s Plus (msm8953) fails to boot with lk2nd – "Error: failed to load kernel"

iamdonut215, 2025-07-04:
Hello,
I’m trying to use lk2nd on my Moto G5s Plus (msm8953), but it fails to boot the kernel. When I power on the device, I get the following error on screen:

Start Up Failed:

Your device didn't start up successfully.
Use the Software Repair Assistant on computer to repair your device.
Connect your device to your computer to get the Software Repair Assistant.

AP Fastboot Flash Mode (Secure)
Error: failed to load kernel!
Fastboot Reason: Fall-through from charger boot mode
USB connected

This issue happens right after flashing lk2nd as the boot

[image]

wonderfulShrineMaidenOfParadise, 2025-07-04:
@playday3008 could you boot lk2nd 20.0 release?

iamdonut215, 2025-07-04:
same errors

crushjz, 2025-07-24:
I have the same issue on my Motorola G5, I've tried with v20 and v19

crushjz, 2025-07-24:
I have the same issue on my Motorola G5, I've tried with v20 and v19. I also built from `main` and same error.

This is the output of `fastboot flash boot qcom-msm8953-lk2nd.img`:

```
(bootloader) is-logical:boot: not found
(bootloader) is-logical:boot: not found
Sending 'boot' (324 KB)                            OKAY [  0.008s]
Writing 'boot'                                     (bootloader) Image not signed or corrupt
OKAY [  0.039s]
Finished. Total time: 0.385s
```

barni2000, 2025-07-24:
> I have the same issue on my Motorola G5, I've tried with v20 and v19. I also built from `main` and same error.
>
> This is the output of `fastboot flash boot qcom-msm8953-lk2nd.img`:
>
> ```
> (bootloader) [...]

If you build it it should work, `make TOOLCHAIN_PREFIX=arm-none-eabi- lk2nd-msm8953`.