msm8916-mainline/lk2nd #563: lk2nd cannot boot (or automatically reset) on msm8956-sony-loire-kugo (Xperia X Compact)

Icenowy, 2025-07-06:
I tried to build a lk2nd-msm8952 and run it with `fastboot boot build-lk2nd-msm8952/lk2nd.img` on my Docomo Sony Xperia X Compact, the fastboot command succeeded but the phone then reboots itself to normal OS.

My phone's serial number does not start with BH902H as expected in `lk2nd/device/dts/msm8952/msm8956-mtp.dts`, so I patched it to BH905X (which is what my phone's serial number starts with), but still fails; then I tried to add a * to the end of the match string, still no enhance.

I tried `fastboot flash boot build-lk2nd-msm8952/lk2nd.img` too and it just bricks my phone into bootloop (I can still use vol+ to activate Sony S1 fastboot and flash back the normal boot partition).

barni2000, 2025-07-17:
Are the `msm-id` and `board-id` same like on `BH902H`?

Have you tried `fastboot boot build-k2nd-msm8952/lk2nd.img`?

Maybe you should try adjust the build a bit like
`make LK2ND_ADTBS="msm8956-mtp.dtb" LK2ND_DTBS="" LK2ND_QCDTBS=""`

Icenowy, 2025-07-19:
I verified msm-id and board-id on a living LineageOS system, and it's correct (the boot.img file contains 4 DTB files, all with MTP board-id and the msm-id corresponds to 8956/8956v1.1/8976/897v1.1, and the 8956v1.1 one seems to be the actual used one).

What do you mean by `fastboot boot build-k2nd-msm8952/lk2nd.img`? I tried first `fastboot boot` and then `fastboot flash boot` (trying the latter is because the S1 bootloader seems to omit enabling display when fastboot).

Not tried building an image with only the mtp dtb, will try it.

Icenowy, 2025-07-19:
Built an image with only the MTP dtb, it doesn't work either...

Maybe I should try to let the lk2nd DTB include an true dtb?

Icenowy, 2025-07-19:
Prepend a whole / node from a decompiled mainilne msm8956-sony-loire-kugo.dtb (which is tested to boot) before `#include <lk2nd.dtsi>` fixes nothing...

Icenowy, 2025-07-30:
Unfortunately Sony didn't provide UART position info for X Compact (and I have no ability to access it on a consumer's device)

barni2000, 2025-07-30:
Maybe there is a possibility use ramoops/pstore with lk2nd.
If you can define it to same memory range what is used in twrp you could get the logs with `adb pull /sys/fs/pstore`

Biswa96, 2025-09-24:
The files in platform/ and target/ are not changed for adding new devices. Compare the dts which you are adding in lk2nd with Android's dts.